devendrachaplot/Neural-SLAM #1: error when python main.py -n1 --auto_gpu_config 0 --split val

Zjhzzzzz, 2020-04-14:
I use googlecolab
I obey all the step before and I get an error when I try to test

`python main.py -n1 --auto_gpu_config 0 --split val`
 I get this error

Traceback (most recent call last):
  File "main.py", line 15, in <module>
    from env import make_vec_envs
  File "/content/Neural-SLAM/env/__init__.py", line 3, in <module>
    from .habitat import construct_envs
  File "/content/Neural-SLAM/env/habitat/__init__.py", line 9, in <module>
    from .exploration_env import Exploration_Env
  File "/content/Neural-SLAM/env/habitat/exploration_env.py", line 28, in <module>
    from env.habitat.utils.noisy_actions import CustomActionSpaceConfiguration
  File "/content/Neural-SLAM/env/habitat/utils/noisy_actions.py", line 12, in <module>
    from habitat_sim.agent.controls import register_move_fn
ImportError: cannot import name 'register_move_fn'

I find there is no register_move_fn in habitat-sim/habitat_sim/agent/controls
Troubled with this, can u help me?
thank you!

devendrachaplot, 2020-04-15:
Hi,
This is likely because you are using a different version of habitat-sim and/or habitat-api. Can you confirm if you are checking out the commits mentioned in the README.md for habitat-sim and habitat-api?

shahhaard47, 2020-04-19:
I got stuck at the same place. However the traceback stops at `/content/Neural-SLAM/env/habitat/exploration_env.py`. I verified I was on the specified branches in habitat-sim and habitat-api. 

Following is my traceback:

```
Traceback (most recent call last):
  File "main.py", line 15, in <module>
    from env import make_vec_envs
  File "/Users/haardshah/development/gatech/DL_CS7643/habitat/habitat-submission/neural-SLAM/env/__init__.py", line 3, in <module>
    from .habitat import construct_envs
  File "/Users/haardshah/development/gatech/DL_CS7643/habitat/habitat-submission/neural-SLAM/env/habitat/__init__.py", line 9, in <module>
    from .exploration_env import Exploration_Env
  File "/Users/haardshah/development/gatech/DL_CS7643/habitat/habitat-submission/neural-SLAM/env/habitat/exploration_env.py", line 29, in <module>
    import env.habitat.utils.pose as pu
AttributeError: module 'env' has no attribute 'habitat'
```

shahhaard47, 2020-04-20:
It's probably not the most ideal solution, but I changed the `import` statements to `from import`  statements. i.e.:
```
# import env.habitat.utils.pose as pu
from env.habitat.utils import pose as pu
```
Also I realize that my error may be different from the version issue in the original post.

devendrachaplot, 2020-04-20:
This error is some bug in python 3.6, if you update to python 3.7 it should be fixed without any change in code.
https://stackoverflow.com/questions/24807434/imports-in-init-py-and-import-as-statement

Zjhzzzzz, 2020-04-22:
> This error is some bug in python 3.6, if you update to python 3.7 it should be fixed without any change in code.
> https://stackoverflow.com/questions/24807434/imports-in-init-py-and-import-as-statement

Thank you very much

closed 2020-04-22 by Zjhzzzzz

KenaHemnani, 2020-12-05:
Hello all, I was facing the same error **ImportError: cannot import name 'register_move_fn'** however my habitat-sim and habitat-api setup was running fine.
As suggested I installed everything in the new virtual env with python3.7. Now, I am getting following error

(ANS1) hemnani1@BOSS8-OS:~/DIR/habitat-api$ python examples/example.py        
  Traceback (most recent call last):
  File "examples/example.py", line 7, in <module>
    import habitat
  File "/DATA/hemnani1/DIR/habitat-api/habitat/__init__.py", line 7, in <module>
    from habitat.config import Config, get_config
  File "/DATA/hemnani1/DIR/habitat-api/habitat/config/__init__.py", line 7, in <module>
    from yacs.config import CfgNode as Config
  File "/DATA/hemnani1/.local/lib/python3.7/site-packages/yacs/config.py", line 30, in <module>
    import yaml
**ModuleNotFoundError: No module named 'yaml'**

Now, even the example.py file of habitat is not working .
How did you all resolve this error? and what things you took care while installation . Please, suggest me some solution.

KenaHemnani, 2020-12-06:
After trying in python 3.7:
I am getting following error:

(ANS1) hemnani1@BOSS8-OS:~/DIR/habitat-api$ python examples/example.py
2020-12-06 15:36:06,378 initializing sim Sim-v0
Traceback (most recent call last):
File "examples/example.py", line 25, in
example()
File "examples/example.py", line 11, in example
env = habitat.Env(config=habitat.get_config("configs/tasks/pointnav.yaml"))
File "/DATA/hemnani1/DIR/habitat-api/habitat/core/env.py", line 93, in init
id_sim=self._config.SIMULATOR.TYPE, config=self._config.SIMULATOR
File "/DATA/hemnani1/DIR/habitat-api/habitat/sims/registration.py", line 19, in make_sim
return _sim(**kwargs)
File "/DATA/hemnani1/DIR/habitat-api/habitat/sims/habitat_simulator/init.py ", line 24, in init
raise habitat_sim_import_error
File "/DATA/hemnani1/DIR/habitat-api/habitat/sims/habitat_simulator/init.py ", line 7, in _try_register_habitat_sim
import habitat_sim
File "/DATA/hemnani1/anaconda3/envs/ANS1/lib/python3.7/site-packages/habitat_si m/init.py", line 13, in
from habitat_sim import ( # noqa: F401
File "/DATA/hemnani1/anaconda3/envs/ANS1/lib/python3.7/site-packages/habitat_si m/agent/init.py", line 7, in
from .agent import * # noqa: F401,F403
File "/DATA/hemnani1/anaconda3/envs/ANS1/lib/python3.7/site-packages/habitat_si m/agent/agent.py", line 10, in
import magnum as mn
File "/DATA/hemnani1/.local/lib/python3.7/site-packages/magnum/init.py", li ne 28, in
from _magnum import *
ImportError: Python version mismatch: module was compiled for Python 3.6, but the interpreter version is incompatible: 3.7.9 (default, Aug 31 2020, 12:42:55)

Can someone help me to resolve this?