devendrachaplot/Neural-SLAM #4: Exploration to PointGoal Transfer

rpartsey, 2020-08-17:
Hi dear @devendrachaplot 
Thank you for open-sourcing your work!
I enjoyed reading your paper "Learning To Explore Using Active Neural SLAM". 
The experiment setup and usage instructions are concise and easy to follow. I was able to run the code without any issues (except import errors fixed here https://github.com/devendrachaplot/Neural-SLAM/issues/1). The environment exploration visualizations generated by pre-trained models look amazing!
 
I'm struggling with _Exploration to PointGoal Transfer_. 
Basically, I understand the idea that the _GlobalPolicy_ should be fixed to output  _PointGoal_ coordinates. 
But I fail to implement this idea in code.

* So the **general question** is: How to transform episode _PointGoal_ coordinates to "local map" coordinates so that they could be used by the short-term goal planner?
*I see that _GlobalPolicy_'s output is a point in "local map" coordinates. My intuition is that _PointGoal_ coordinates should be converted to "full map" coordinates and then to "local map" coordinates.

* Could you, please, also explain the math behind this expression? What does `st` mean? Why do we add 180 to orientation?  https://github.com/devendrachaplot/Neural-SLAM/blob/e833e8f20c5a8d71552a5d22fb78ad6980cfc80c/env/habitat/exploration_env.py#L613
* And how `get_grid` function works?
https://github.com/devendrachaplot/Neural-SLAM/blob/e833e8f20c5a8d71552a5d22fb78ad6980cfc80c/utils/model.py#L7
*I understand that it returns rotation and transition matrix. But, again, what is the math behind? I'm also confused with `F.affine_grid` function. And the similar transformation here https://github.com/devendrachaplot/Neural-SLAM/blob/e833e8f20c5a8d71552a5d22fb78ad6980cfc80c/model.py#L206
* And what is the intuition for applying `get_new_pose_batch` function?  https://github.com/devendrachaplot/Neural-SLAM/blob/e833e8f20c5a8d71552a5d22fb78ad6980cfc80c/model.py#L262

rpartsey, 2020-08-18:
I also wonder how you interpret `alpha_beta_gamma` angles returned by the `as_euler_angles` function  (I mean angle - axis correspondence, which angle corresponds to which axis rotation?) and what is the intuition behind following `if` statement  https://github.com/devendrachaplot/Neural-SLAM/blob/d5bc7886a4fdccae0287b7d0a26bf3800126b563/env/habitat/exploration_env.py#L375

rpartsey, 2020-08-26:
Thanks a lot!
The agent reaches the PointGoal as expected.

I've also added `config_env.TASK.SENSORS.append("POINTGOAL_SENSOR")` in `Neural-SLAM/env/habitat/__init__.py` after line 52.
https://github.com/devendrachaplot/Neural-SLAM/blob/e833e8f20c5a8d71552a5d22fb78ad6980cfc80c/env/habitat/__init__.py#L52

The only thing that I'd like to clarify is how to properly handle the case when the agent reaches the goal in less than `args.max_episode_length`. 

Currently, after the `STOP` action is called `_episode_over` property is set to `True` and if the `step` is less than `args.max_episode_length`, then the next `envs.step(l_action)` will be called on the finished episode and this results in `AssertionError: Episode over, call reset before calling step`.

devendrachaplot, 2020-08-26:
The hacky/easy way of getting rid of this error is just waiting till the max_episode_length in the step function:

You can add the following at L198 in env/habitat/exploration_env.py:
```
if self. _previous_action == 0:
    if self.timestep >= args.max_episode_length:
          done = True
    else:
          done = False
    null_state = np.zeros((3, self.args.frame_height, self.args.frame_width))
    return null_state, 0., done, self.info
```

The better way, which will require a lot of code change, is to pass is to set done=True when self._previous_action == 0 in step function and then checking which thread has episode finished in main.py (at L371) and resetting map for a thread when it is finished. You will also need to sync this with local steps (because map get shifted between global steps). This should not change the results, it will only help in reducing the runtime.

Some more notes:
- If you want to run the Pointgoal task with GPS+Compass (without pose noise), you need to run main.py with `--noisy_actions 0`, `--noisy_odometry 0` and `--use_pose_estimation 0` arguments
- The pose estimator is trained with our pose noise models based on real-world data, it is not expected to work with the pose noise models implemented in Habitat currently.
- To get the best results during evaluation, run with `--map_size_cm 3000` and `--global_downscaling 1` arguments. This will increase the runtime slightly.

rpartsey, 2020-08-26:
@devendrachaplot Thank you!

closed 2020-08-26 by rpartsey