devendrachaplot/Neural-SLAM #6: Error running main.py "BrokenPipeError: [Errno 32] Broken pipe"

GhadeerElmkaiel, 2020-09-16:
**When I run the following code I get an error:**

_python main.py --split val --eval 1 --train_global 0 --train_local 0 --train_slam 0 --load_global pretrained_models/model_best.global --load_local pretrained_models/model_best.local --load_slam pretrained_models/model_best.slam --no_cuda --task_config tasks/pointnav_mp3d.yaml --num_processes 1_

**The output is as follows:**

 Dumping at ./tmp//models/exp1/
Namespace(alpha=0.99, auto_gpu_config=1, camera_height=1.25, clip_param=0.2, collision_threshold=0.2, cuda=False, du_scale=2, dump_location='./tmp/', entropy_coef=0.001, env_frame_height=256, env_frame_width=256, eps=1e-05, eval=1, exp_loss_coeff=1.0, exp_name='exp1', frame_height=128, frame_width=128, gamma=0.99, global_downscaling=2, global_hidden_size=256, global_lr=2.5e-05, goals_size=2, hfov=90.0, load_global='pretrained_models/model_best.global', load_local='pretrained_models/model_best.local', load_slam='pretrained_models/model_best.slam', local_hidden_size=512, local_optimizer='adam,lr=0.0001', local_policy_update_freq=5, log_interval=10, map_pred_threshold=0.5, map_resolution=5, map_size_cm=2400, max_episode_length=1000, max_grad_norm=0.5, no_cuda=True, noise_level=1.0, noisy_actions=1, noisy_odometry=1, num_episodes=1000000, num_global_steps=40, num_local_steps=25, num_mini_batch=0, num_processes=1, num_processes_on_first_gpu=0, num_processes_per_gpu=11, obs_threshold=1, obstacle_boundary=5, pose_loss_coeff=10000.0, ppo_epoch=4, pretrained_resnet=1, print_images=0, proj_loss_coeff=1.0, randomize_env_every=1000, save_interval=1, save_periodic=500000, save_trajectory_data='0', seed=1, short_goal_dist=1, sim_gpu_id=0, slam_batch_size=72, slam_iterations=10, slam_memory_size=500000, slam_optimizer='adam,lr=0.0001', split='val', task_config='tasks/pointnav_mp3d.yaml', tau=0.95, total_num_scenes='auto', train_global=0, train_local=0, train_slam=0, use_deterministic_local=0, use_gae=False, use_pose_estimation=2, use_recurrent_global=0, use_recurrent_local=1, value_loss_coef=0.5, vis_type=1, vision_range=64, visualize=0)
Loading data/scene_datasets/mp3d/2azQ1b91cZZ/2azQ1b91cZZ.glb
2020-09-16 17:57:23,377 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0916 17:57:23.383668  5564 WindowlessContext.cpp:98] [EGL] Detected 4 EGL devices
I0916 17:57:23.383796  5564 WindowlessContext.cpp:119] [EGL] Selected EGL device 0 for CUDA device 0
I0916 17:57:23.385957  5564 WindowlessContext.cpp:133] [EGL] Version: 1.5
I0916 17:57:23.385967  5564 WindowlessContext.cpp:134] [EGL] Vendor: NVIDIA
Renderer: GeForce GTX 1060 6GB/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 440.100
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    nv-egl-incorrect-gl11-function-pointers
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
Trade::AbstractImporter::image2D(): index out of range
Traceback (most recent call last):
  File "main.py", line 769, in <module>
    main()
  File "main.py", line 119, in main
    envs = make_vec_envs(args)
  File "/home/ubuntu/Projects/Active-Neural-SLAM/Neural-SLAM/Neural-SLAM/env/__init__.py", line 7, in make_vec_envs
    envs = construct_envs(args)
  File "/home/ubuntu/Projects/Active-Neural-SLAM/Neural-SLAM/Neural-SLAM/env/habitat/__init__.py", line 102, in construct_envs
    range(args.num_processes))
  File "/home/ubuntu/Projects/Active-Neural-SLAM/Neural-SLAM/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 117, in __init__
    read_fn() for read_fn in self._connection_read_fns
  File "/home/ubuntu/Projects/Active-Neural-SLAM/Neural-SLAM/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 117, in <listcomp>
    read_fn() for read_fn in self._connection_read_fns
  File "/home/ubuntu/anaconda3/envs/ns/lib/python3.7/multiprocessing/connection.py", line 250, in recv
    buf = self._recv_bytes()
  File "/home/ubuntu/anaconda3/envs/ns/lib/python3.7/multiprocessing/connection.py", line 407, in _recv_bytes
    buf = self._recv(4)
  File "/home/ubuntu/anaconda3/envs/ns/lib/python3.7/multiprocessing/connection.py", line 379, in _recv
    chunk = read(handle, remaining)
ConnectionResetError: [Errno 104] Connection reset by peer
Exception ignored in: <function VectorEnv.__del__ at 0x7f90ceac2488>
Traceback (most recent call last):
  File "/home/ubuntu/Projects/Active-Neural-SLAM/Neural-SLAM/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 487, in __del__
    self.close()
  File "/home/ubuntu/Projects/Active-Neural-SLAM/Neural-SLAM/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 351, in close
    write_fn((CLOSE_COMMAND, None))
  File "/home/ubuntu/anaconda3/envs/ns/lib/python3.7/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/home/ubuntu/anaconda3/envs/ns/lib/python3.7/multiprocessing/connection.py", line 404, in _send_bytes
    self._send(header + buf)
  File "/home/ubuntu/anaconda3/envs/ns/lib/python3.7/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
BrokenPipeError: [Errno 32] Broken pipe

__________________________________________________________________
Note: I am using different task config because I didn't download the Gibson dataset yet.
It would be great if you can help me with this error

devendrachaplot, 2020-09-17:
Hi,
This seems like an issue with habitat installation. Quick way to check this is by running examples/benchmark.py in habitat-api directory (where you installed habitat-api, not the submodule within Neural-SLAM directory). If it throws an error, it indicates habitat-sim or api is not installed correctly.

GhadeerElmkaiel, 2020-09-18:
Hi again,
thanks for the fast response.
I have tried to run benchmark.py  code and it works fine.
__________________________________________________________________________
python examples/benchmark.py 
2020-09-18 11:01:55,421 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0918 11:01:55.437364 25320 WindowlessContext.cpp:98] [EGL] Detected 4 EGL devices
I0918 11:01:55.450826 25320 WindowlessContext.cpp:119] [EGL] Selected EGL device 0 for CUDA device 0
I0918 11:01:55.452826 25320 WindowlessContext.cpp:133] [EGL] Version: 1.5
I0918 11:01:55.452836 25320 WindowlessContext.cpp:134] [EGL] Vendor: NVIDIA
Renderer: GeForce GTX 1060 6GB/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 440.100
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    nv-egl-incorrect-gl11-function-pointers
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
I0918 11:01:56.511303 25320 simulator.py:80] Loaded navmesh data/scene_datasets/habitat-test-scenes/skokloster-castle.navmesh
2020-09-18 11:01:56,512 initializing task Nav-v0
spl: 0.000
I0918 11:02:03.061357 25320 Simulator.cpp:42] Deconstructing Simulator
I0918 11:02:03.061380 25320 SemanticScene.h:40] Deconstructing SemanticScene
I0918 11:02:03.061388 25320 SceneManager.h:24] Deconstructing SceneManager
I0918 11:02:03.061393 25320 SceneGraph.h:20] Deconstructing SceneGraph
I0918 11:02:03.068886 25320 Renderer.cpp:38] Deconstructing Renderer
I0918 11:02:03.068941 25320 WindowlessContext.cpp:240] Deconstructing GL context
I0918 11:02:03.068997 25320 PathFinder.h:113] Deconstructing PathFinder

I also wanted to check habitat-sim but it seems that this branch does not have /build/viewer to test the simulator.

devendrachaplot, 2020-09-21:
I am unsure what the error is. The error is not related to the model code, but happening when starting the environments and loading the scenes. I would suggest trying with the Gibson dataset.

devendrachaplot, 2020-10-27:
I am closing the issue for now due to inactivity, please post if you found the solution to the problem.

closed 2020-10-27 by devendrachaplot

CloudWave818, 2020-11-03:
> **When I run the following code I get an error:**
> 
> _python main.py --split val --eval 1 --train_global 0 --train_local 0 --train_slam 0 --load_global pretrained_models/model_best.global --load_local pretrained_models/model_best.local --load_slam pretrained_models/model_best.slam --no_cuda --task_config tasks/pointnav_mp3d.yaml --num_processes 1_
> 
> **The output is as follows:**
> 
> Dumping at ./tmp//models/exp1/
> Namespace(alpha=0.99, auto_gpu_config=1, camera_height=1.25, clip_param=0.2, collision_threshold=0.2, cuda=False, du_scale=2, dump_location='./tmp/', entropy_coef=0.001, env_frame_height=256, env_frame_width=256, eps=1e-05, eval=1, exp_loss_coeff=1.0, exp_name='exp1', frame_height=128, frame_width=128, gamma=0.99, global_downscaling=2, global_hidden_size=256, global_lr=2.5e-05, goals_size=2, hfov=90.0, load_global='pretrained_models/model_best.global', load_local='pretrained_models/model_best.local', load_slam='pretrained_models/model_best.slam', local_hidden_size=512, local_optimizer='adam,lr=0.0001', local_policy_update_freq=5, log_interval=10, map_pred_threshold=0.5, map_resolution=5, map_size_cm=2400, max_episode_length=1000, max_grad_norm=0.5, no_cuda=True, noise_level=1.0, noisy_actions=1, noisy_odometry=1, num_episodes=1000000, num_global_steps=40, num_local_steps=25, num_mini_batch=0, num_processes=1, num_processes_on_first_gpu=0, num_processes_per_gpu=11, obs_threshold=1, obstacle_boundary=5, pose_loss_coeff=10000.0, ppo_epoch=4, pretrained_resnet=1, print_images=0, proj_loss_coeff=1.0, randomize_env_every=1000, save_interval=1, save_periodic=500000, save_trajectory_data='0', seed=1, short_goal_dist=1, sim_gpu_id=0, slam_batch_size=72, slam_iterations=10, slam_memory_size=500000, slam_optimizer='adam,lr=0.0001', split='val', task_config='tasks/pointnav_mp3d.yaml', tau=0.95, total_num_scenes='auto', train_global=0, train_local=0, train_slam=0, use_deterministic_local=0, use_gae=False, use_pose_estimation=2, use_recurrent_global=0, use_recurrent_local=1, value_loss_coef=0.5, vis_type=1, vision_range=64, visualize=0)
> Loading data/scene_datasets/mp3d/2azQ1b91cZZ/2azQ1b91cZZ.glb
> 2020-09-16 17:57:23,377 initializing sim Sim-v0
> WARNING: Logging before InitGoogleLogging() is written to STDERR
> I0916 17:57:23.383668 5564 WindowlessContext.cpp:98] [EGL] Detected 4 EGL devices
> I0916 17:57:23.383796 5564 WindowlessContext.cpp:119] [EGL] Selected EGL device 0 for CUDA device 0
> I0916 17:57:23.385957 5564 WindowlessContext.cpp:133] [EGL] Version: 1.5
> I0916 17:57:23.385967 5564 WindowlessContext.cpp:134] [EGL] Vendor: NVIDIA
> Renderer: GeForce GTX 1060 6GB/PCIe/SSE2 by NVIDIA Corporation
> OpenGL version: 4.6.0 NVIDIA 440.100
> Using optional features:
> GL_ARB_ES2_compatibility
> GL_ARB_direct_state_access
> GL_ARB_get_texture_sub_image
> GL_ARB_invalidate_subdata
> GL_ARB_multi_bind
> GL_ARB_robustness
> GL_ARB_separate_shader_objects
> GL_ARB_texture_filter_anisotropic
> GL_ARB_texture_storage
> GL_ARB_texture_storage_multisample
> GL_ARB_vertex_array_object
> GL_KHR_debug
> Using driver workarounds:
> nv-egl-incorrect-gl11-function-pointers
> no-layout-qualifiers-on-old-glsl
> nv-zero-context-profile-mask
> nv-implementation-color-read-format-dsa-broken
> nv-cubemap-inconsistent-compressed-image-size
> nv-cubemap-broken-full-compressed-image-query
> nv-compressed-block-size-in-bits
> Trade::AbstractImporter::image2D(): index out of range
> Traceback (most recent call last):
> File "main.py", line 769, in 
> main()
> File "main.py", line 119, in main
> envs = make_vec_envs(args)
> File "/home/ubuntu/Projects/Active-Neural-SLAM/Neural-SLAM/Neural-SLAM/env/**init**.py", line 7, in make_vec_envs
> envs = construct_envs(args)
> File "/home/ubuntu/Projects/Active-Neural-SLAM/Neural-SLAM/Neural-SLAM/env/habitat/**init**.py", line 102, in construct_envs
> range(args.num_processes))
> File "/home/ubuntu/Projects/Active-Neural-SLAM/Neural-SLAM/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 117, in **init**
> read_fn() for read_fn in self._connection_read_fns
> File "/home/ubuntu/Projects/Active-Neural-SLAM/Neural-SLAM/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 117, in 
> read_fn() for read_fn in self._connection_read_fns
> File "/home/ubuntu/anaconda3/envs/ns/lib/python3.7/multiprocessing/connection.py", line 250, in recv
> buf = self._recv_bytes()
> File "/home/ubuntu/anaconda3/envs/ns/lib/python3.7/multiprocessing/connection.py", line 407, in _recv_bytes
> buf = self._recv(4)
> File "/home/ubuntu/anaconda3/envs/ns/lib/python3.7/multiprocessing/connection.py", line 379, in _recv
> chunk = read(handle, remaining)
> ConnectionResetError: [Errno 104] Connection reset by peer
> Exception ignored in: <function VectorEnv.**del** at 0x7f90ceac2488>
> Traceback (most recent call last):
> File "/home/ubuntu/Projects/Active-Neural-SLAM/Neural-SLAM/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 487, in **del**
> self.close()
> File "/home/ubuntu/Projects/Active-Neural-SLAM/Neural-SLAM/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 351, in close
> write_fn((CLOSE_COMMAND, None))
> File "/home/ubuntu/anaconda3/envs/ns/lib/python3.7/multiprocessing/connection.py", line 206, in send
> self._send_bytes(_ForkingPickler.dumps(obj))
> File "/home/ubuntu/anaconda3/envs/ns/lib/python3.7/multiprocessing/connection.py", line 404, in _send_bytes
> self._send(header + buf)
> File "/home/ubuntu/anaconda3/envs/ns/lib/python3.7/multiprocessing/connection.py", line 368, in _send
> n = write(self._handle, buf)
> BrokenPipeError: [Errno 32] Broken pipe
> 
> Note: I am using different task config because I didn't download the Gibson dataset yet.
> It would be great if you can help me with this error

I encountered the same problem when using the MP3D dataset. Did you solve it?