devendrachaplot/Neural-SLAM #12: No module named 'gym.spaces.dict_space'

Sunnyzhr, 2020-10-25:
When running `(habitat) root@c:~/Desktop/Neural-SLAM# python main.py -n1 --auto_gpu_config 0 --split val`,  I get error:
>Traceback (most recent call last):
  File "main.py", line 12, in <module>
    import gym
ModuleNotFoundError: No module named 'gym'

Thus, I try to install gym via conda `(habitat) root@c:~/Desktop/Neural-SLAM# conda install -c conda-forge gym `. (The version of gym is 0.17.3) Then I run `(habitat) root@c:~/Desktop/Neural-SLAM# python main.py -n1 --auto_gpu_config 0 --split val` again, and get error:

> Traceback (most recent call last):
  File "main.py", line 15, in <module>
    from env import make_vec_envs
  File "/home/u/Desktop/Neural-SLAM/env/__init__.py", line 3, in <module>
    from .habitat import construct_envs
  File "/home/u/Desktop/Neural-SLAM/env/habitat/__init__.py", line 6, in <module>
    from habitat.config.default import get_config as cfg_env
  File "/home/u/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat/__init__.py", line 8, in <module>
    from habitat.core.agent import Agent
  File "/home/u/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat/core/agent.py", line 11, in <module>
    from habitat.core.simulator import Observations
  File "/home/u/anaconda3/envs/habitat/lib/python3.6/site-packages/habitat/core/simulator.py", line 14, in <module>
    from gym.spaces.dict_space import Dict as SpaceDict
ModuleNotFoundError: No module named 'gym.spaces.dict_space'

## Expectation
Successfully run the code.

Sunnyzhr, 2020-10-25:
I delete all files, and re-download the entire folder.  When running `(habitat) root@c:~/Desktop/Neural-SLAM# python main.py -n1 --auto_gpu_config 0 --split val
` again, I get different error:

> Traceback (most recent call last):
  File "main.py", line 15, in <module>
    from env import make_vec_envs
  File "/home/u/Desktop/Neural-SLAM/env/__init__.py", line 3, in <module>
    from .habitat import construct_envs
  File "/home/u/Desktop/Neural-SLAM/env/habitat/__init__.py", line 9, in <module>
    from .exploration_env import Exploration_Env
  File "/home/u/Desktop/Neural-SLAM/env/habitat/exploration_env.py", line 14, in <module>
    from torchvision import transforms
ModuleNotFoundError: No module named 'torchvision'

This time, ModuleNotFoundError changes from 'gym' to 'torchvision'. Then I try to `(habitat) root@c:~/Desktop/Neural-SLAM# conda install torchvision
` and run `(habitat) root@c:~/Desktop/Neural-SLAM# python main.py -n1 --auto_gpu_config 0 --split val
` again. I get the error below:

> ModuleNotFoundError: No module named 'habitat.sims.habitat_simulator.action_spaces'

How to solve these ModuleNotFoundErrors?

devendrachaplot, 2020-10-26:
Hi,
You need to install specific older versions of habitat-sim and habitat-api for running the Active Neural SLAM code. Check the instructions on installing dependencies in the README for specific versions:

https://github.com/devendrachaplot/Neural-SLAM/blob/master/README.md#installing-dependencies

Sunnyzhr, 2020-10-27:
@devendrachaplot Thanks for reply!
I follow the instructions, but still get a little trouble:

# My Steps (before Error)
0. Preparations
Intructed by [BuildFromSource](https://github.com/facebookresearch/habitat-sim/blob/master/BUILD_FROM_SOURCE.md)
```
sudo -s
conda create -n habitat python=3.6 cmake=3.14.0
conda activate habitat
sudo apt-get update || true
sudo apt-get install -y --no-install-recommends \
     libjpeg-dev libglm-dev libgl1-mesa-glx libegl1-mesa-dev mesa-utils xorg-dev freeglut3-dev
```
1. Installing habitat-sim
Instructed by [ActiveNeuralSLAM](https://github.com/devendrachaplot/Neural-SLAM)
```
cd ~/Desktop
git clone https://github.com/facebookresearch/habitat-sim.git
cd habitat-sim; git checkout 9575dcd45fe6f55d2a44043833af08972a7895a9; 
```
**The checkout information** is as below:

> (habitat) root@c:~/Desktop# cd habitat-sim; git checkout 9575dcd45fe6f55d2a44043833af08972a7895a9; 
> Note: checking out '9575dcd45fe6f55d2a44043833af08972a7895a9'.
> 
> You are in 'detached HEAD' state. You can look around, make experimental
> changes and commit them, and you can discard any commits you make in this
> state without impacting any branches by performing another checkout.
> 
> If you want to create a new branch to retain commits you create, you may
> do so (now or later) by using -b with the checkout command again. Example:
> 
>   git checkout -b <new-branch-name>
> 
> HEAD is now at 9575dcd Bullet Physics Integration #1 (#91)

```
pip install -r requirements.txt; 
```
I get **a warning**:

> (habitat) root@c:~/Desktop/habitat-sim# pip install -r requirements.txt; 
> **WARNING**: The directory '/home/u/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
> 

```
python setup.py install --headless
```
I get the same **warning**:
> Using /home/u/anaconda3/envs/habitat/lib/python3.6/site-packages
> Finished processing dependencies for habitat-sim==0.1.2
> **WARNING:** The directory '/home/u/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
> Processing ./build/deps/magnum-bindings/src/python
> Building wheels for collected packages: magnum
>   Building wheel for magnum (setup.py) ... done
>   Created wheel for magnum: filename=magnum-0.0.0-cp36-cp36m-linux_x86_64.whl size=1310443 sha256=d5590e83d54bc3a439f78670780bc949dc9fde561942b2aec0de69a5bc732bb7
>   Stored in directory: /tmp/pip-ephem-wheel-cache-q697ojl2/wheels/31/f6/f6/24e2dea66d597afa365ec31feb5ef879b12b7e448712ffaa0d
> Successfully built magnum
> Installing collected packages: magnum
> Successfully installed magnum-0.0.0

2. Installing habitat-api:
Instructed by [ActiveNeuralSLAM](https://github.com/devendrachaplot/Neural-SLAM)
```
cd ~/Desktop
git clone https://github.com/facebookresearch/habitat-api.git
cd habitat-api; git checkout b5f2b00a25627ecb52b43b13ea96b05998d9a121; 
```
The checkout information is as below:
[... 98 lines cut ...]
>   File "/home/u/Desktop/habitat-sim/habitat_sim/agent/agent.py", line 14, in <module>
>     import habitat_sim.bindings as hsim
>   File "/home/u/Desktop/habitat-sim/habitat_sim/bindings/__init__.py", line 24, in <module>
>     from habitat_sim._ext.habitat_sim_bindings import Simulator as SimulatorBackend
> ModuleNotFoundError: No module named 'habitat_sim._ext'
> 
Situation-3: If I write `~/.bashrc` without any extra code:
> (habitat) root@c:~/Desktop/Neural-SLAM# python main.py -n1 --auto_gpu_config 0 --split val
> Traceback (most recent call last):
>   File "main.py", line 15, in <module>
>     from env import make_vec_envs
>   File "/home/u/Desktop/Neural-SLAM/env/__init__.py", line 3, in <module>
>     from .habitat import construct_envs
>   File "/home/u/Desktop/Neural-SLAM/env/habitat/__init__.py", line 9, in <module>
>     from .exploration_env import Exploration_Env
>   File "/home/u/Desktop/Neural-SLAM/env/habitat/exploration_env.py", line 29, in <module>
>     import env.habitat.utils.pose as pu
> AttributeError: module 'env' has no attribute 'habitat'
> 

So, what should I write in `~/.bashrc`? And how to fix the problem? 

**Thank you very much for reading!**
# Expectation
Successfully run the codes.

Sunnyzhr, 2020-10-27:
@devendrachaplot Thank you so much for reply!

I follow your instructions carefully, and encounter another error`AttributeError: module 'env' has no attribute 'habitat'`. Fortunately, I follow [your instruction](https://github.com/devendrachaplot/Neural-SLAM/issues/1) again to solve the problem. (By change `import ...` to `from ... import ...` ) THANKS again!!!

After that, I encounter another trouble that I cannot run with GPU. I fix it by change my torch(cuda11.0) to torch(cuda10.0).

Then, I start to run commands of the tutorial, I've encounter several new problems as below:
## New problem 1 - `python main.py -n1 --auto_gpu_config 0 --split val`
The command runs without error at the first 38 minuets. However, then it encounter errors as below. Additionally, I find that there are some other [people who encouter the same error](https://blog.csdn.net/xyt723916/article/details/109284565). **The code will fail at `Cantwell.glb`**
> ...
> ...
> Time: 00d 00h 37m 49s, num timesteps 980, FPS 0,
> 	Rewards:
> 	Losses: Local Loss: 4.630, SLAM Loss proj/exp/pose:0.0004/0.0004/0.0101
> Time: 00d 00h 38m 11s, num timesteps 990, FPS 0,
> 	Rewards:
> 	Losses: Local Loss: 4.661, SLAM Loss proj/exp/pose:0.0003/0.0003/0.0100
> 2020-10-27 17:39:42,208 **Computing map for data/scene_datasets/gibson/Cantwell.glb**
> Traceback (most recent call last):
>   File "main.py", line 769, in <module>
>     main()
>   File "main.py", line 617, in main
>     g_agent.update(g_rollouts)
>   File "/home/u/Desktop/Neural-SLAM/algo/ppo.py", line 58, in update
>     for sample in data_generator:
>   File "/home/u/Desktop/Neural-SLAM/utils/storage.py", line 95, in feed_forward_generator
>     mini_batch_size = batch_size // num_mini_batch
> ZeroDivisionError: integer division or modulo by zero
> Exception ignored in: <bound method VectorEnv.__del__ of <env.habitat.habitat_api.habitat.core.vector_env.VectorEnv object at 0x7faf77d21c18>>
> Traceback (most recent call last):
>   File "/home/u/Desktop/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 487, in __del__
>     self.close()
>   File "/home/u/Desktop/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 351, in close
>     write_fn((CLOSE_COMMAND, None))
>   File "/home/u/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 206, in send
>     self._send_bytes(_ForkingPickler.dumps(obj))
>   File "/home/u/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 404, in _send_bytes
>     self._send(header + buf)
>   File "/home/u/anaconda3/envs/habitat/lib/python3.6/multiprocessing/connection.py", line 368, in _send
>     n = write(self._handle, buf)
> BrokenPipeError: [Errno 32] Broken pipe

## New problem 2 - `python main.py`
The error information is:

> (habitat) root@c:~/Desktop/Neural-SLAM# python main.py
> Traceback (most recent call last):
>   File "main.py", line 33, in <module>
>     args = get_args()
>   File "/home/u/Desktop/Neural-SLAM/arguments.py", line 198, in get_args
>     /1024/1024/1024 > 10.0, "Insufficient GPU memory"
> AssertionError: Insufficient GPU memory

I check my GPU memory (I only have 1 GPU) with `torch.cuda.get_device_properties(0).total_memory/1024/1024/1024` . The result is `
 3.81195068359375` which is less than 10. 
I see the [arguments.py](https://github.com/devendrachaplot/Neural-SLAM/blob/master/arguments.py), there are two lines: 
> num_processes_per_gpu = int(gpu_memory/1.4)
> num_processes_on_first_gpu = int((gpu_memory - 10.0)/1.4)

Therefore, **could you please recommend me a value of `num_processes_per_gpu` and `num_processes_on_first_gpu`?**

# Expectation
Successfully run the codes.

Thanks for your attention!

closed 2020-11-22 by devendrachaplot