devendrachaplot/Neural-SLAM #17: Error running main.py "BrokenPipeError: [Errno 32] Broken pipe" with gibson dataset

maomz, 2020-11-20:
Hello,
I encounter the as error as #6  but with gibson dataset. I have also run the benchmark.py code and it works fine.

```
python main.py --split val --eval 1 --train_global 0 --train_local 0 --train_slam 0 \
--load_global pretrained_models/model_best.global \
--load_local pretrained_models/model_best.local \
--load_slam pretrained_models/model_best.slam
```

and I got the following output:
```

Auto GPU config:
Number of processes: 1
Number of processes on GPU 0: 0
Number of processes per GPU: 1
Dumping at ./tmp//models/exp1/
Namespace(alpha=0.99, auto_gpu_config=1, camera_height=1.25, clip_param=0.2, collision_threshold=0.2, cuda=True, du_scale=2, dump_location=‘./tmp/’, entropy_coef=0.001, env_frame_height=256, env_frame_width=256, eps=1e-05, eval=1, exp_loss_coeff=1.0, exp_name=‘exp1’, frame_height=128, frame_width=128, gamma=0.99, global_downscaling=2, global_hidden_size=256, global_lr=2.5e-05, goals_size=2, hfov=90.0, load_global=‘pretrained_models/model_best.global’, load_local=‘pretrained_models/model_best.local’, load_slam=‘pretrained_models/model_best.slam’, local_hidden_size=512, local_optimizer=‘adam,lr=0.0001’, local_policy_update_freq=5, log_interval=10, map_pred_threshold=0.5, map_resolution=5, map_size_cm=2400, max_episode_length=1000, max_grad_norm=0.5, no_cuda=False, noise_level=1.0, noisy_actions=1, noisy_odometry=1, num_episodes=1000000, num_global_steps=40, num_local_steps=25, num_mini_batch=0, num_processes=1, num_processes_on_first_gpu=0, num_processes_per_gpu=1, obs_threshold=1, obstacle_boundary=5, pose_loss_coeff=10000.0, ppo_epoch=4, pretrained_resnet=1, print_images=0, proj_loss_coeff=1.0, randomize_env_every=1000, save_interval=1, save_periodic=500000, save_trajectory_data=‘0’, seed=1, short_goal_dist=1, sim_gpu_id=1, slam_batch_size=72, slam_iterations=10, slam_memory_size=500000, slam_optimizer=‘adam,lr=0.0001’, split=‘val’, task_config=‘tasks/pointnav_gibson.yaml’, tau=0.95, total_num_scenes=1, train_global=0, train_local=0, train_slam=0, use_deterministic_local=0, use_gae=False, use_pose_estimation=2, use_recurrent_global=0, use_recurrent_local=1, value_loss_coef=0.5, vis_type=1, vision_range=64, visualize=0)
Auto GPU config:
Number of processes: 1
Number of processes on GPU 0: 0
Number of processes per GPU: 1
Loading data/scene_datasets/gibson/Cantwell.glb
2020-11-20 16:06:58,655 initializing sim Sim-v0
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1120 16:06:58.661386 17580 WindowlessContext.cpp:98] [EGL] Detected 5 EGL devices
I1120 16:06:58.661504 17580 WindowlessContext.cpp:119] [EGL] Selected EGL device 1 for CUDA device 1
I1120 16:06:58.668072 17580 WindowlessContext.cpp:133] [EGL] Version: 1.5
I1120 16:06:58.668079 17580 WindowlessContext.cpp:134] [EGL] Vendor: NVIDIA
Renderer: GeForce RTX 3090/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 455.45.01
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    nv-egl-incorrect-gl11-function-pointers
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
I1120 16:07:01.149317 17580 simulator.py:80] Loaded navmesh data/scene_datasets/gibson/Cantwell.navmesh
2020-11-20 16:07:01,149 initializing task Nav-v0
2020-11-20 16:07:01,155 Computing map for data/scene_datasets/gibson/Cantwell.glb
Loading slam pretrained_models/model_best.slam
Loading global pretrained_models/model_best.global
Loading local pretrained_models/model_best.local
Traceback (most recent call last):
  File “main.py”, line 769, in <module>
    main()
  File “main.py”, line 273, in main
    local_map[:, 1, :, :], local_pose)
  File “/home/mzmao/ENTER/envs/env01/lib/python3.7/site-packages/torch/nn/modules/module.py”, line 547, in call
    result = self.forward(*input, **kwargs)
  File “/home/mzmao/projects/habitat-sim/habitat-api/Neural-SLAM/model.py”, line 162, in forward
    resnet_output = self.resnet_l5(obs[:, :3, :, :])
  File “/home/mzmao/ENTER/envs/env01/lib/python3.7/site-packages/torch/nn/modules/module.py”, line 547, in call
    result = self.forward(*input, **kwargs)
  File “/home/mzmao/ENTER/envs/env01/lib/python3.7/site-packages/torch/nn/modules/container.py”, line 92, in forward
    input = module(input)
  File “/home/mzmao/ENTER/envs/env01/lib/python3.7/site-packages/torch/nn/modules/module.py”, line 547, in call
    result = self.forward(*input, **kwargs)
  File “/home/mzmao/ENTER/envs/env01/lib/python3.7/site-packages/torch/nn/modules/activation.py”, line 94, in forward
    return F.relu(input, inplace=self.inplace)
  File “/home/mzmao/ENTER/envs/env01/lib/python3.7/site-packages/torch/nn/functional.py”, line 911, in relu
    result = torch.relu_(input)
RuntimeError: CUDA error: no kernel image is available for execution on the device
Exception ignored in: <function VectorEnv.del at 0x7fbc8db51048>
Traceback (most recent call last):
  File “/home/mzmao/projects/habitat-sim/habitat-api/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py”, line 487, in del
    self.close()
  File “/home/mzmao/projects/habitat-sim/habitat-api/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py”, line 351, in close
    write_fn((CLOSE_COMMAND, None))
  File “/home/mzmao/ENTER/envs/env01/lib/python3.7/multiprocessing/connection.py”, line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File “/home/mzmao/ENTER/envs/env01/lib/python3.7/multiprocessing/connection.py”, line 404, in _send_bytes
    self._send(header + buf)
  File “/home/mzmao/ENTER/envs/env01/lib/python3.7/multiprocessing/connection.py”, line 368, in _send
    n = write(self._handle, buf)
BrokenPipeError: [Errno 32] Broken pipe
```

would appreciate for some help,
thanks!

devendrachaplot, 2020-11-22:
Hi,

This looks like a CUDA error, which indicates either pytorch or cuda drivers are not installed correctly. It is unlikely that this error is due to the Neural-SLAM code.

maomz, 2020-11-24:
thanks for the reply,

I fixed this problem by reinstalling the latest version of pytorch.

closed 2020-11-25 by devendrachaplot