devendrachaplot/Neural-SLAM #18: No module named 'attr'

KenaHemnani, 2020-12-02:
After running the following command:
python main.py -n1 --auto_gpu_config 0 --split val

I am encountering following error:

Traceback (most recent call last):
  File "main.py", line 15, in <module>
    from env import make_vec_envs
  File "/DATA/hemnani1/DIR/Neural-SLAM/env/__init__.py", line 3, in <module>
    from .habitat import construct_envs
  File "/DATA/hemnani1/DIR/Neural-SLAM/env/habitat/__init__.py", line 6, in <module>
    from habitat.config.default import get_config as cfg_env
  File "/DATA/hemnani1/DIR/habitat-api/habitat/__init__.py", line 8, in <module>
    from habitat.core.agent import Agent
  File "/DATA/hemnani1/DIR/habitat-api/habitat/core/agent.py", line 11, in <module>
    from habitat.core.simulator import Observations
  File "/DATA/hemnani1/DIR/habitat-api/habitat/core/simulator.py", line 12, in <module>
    import attr
ModuleNotFoundError: No module named 'attr'

It will be very helpful if someone can suggest the possible solution.

KenaHemnani, 2020-12-03:
Also, I did not found this module in habitat-sim or api . Is it a customized module of habitat-sim or one given here https://pypi.org/project/attr/  ?

GhadeerElmkaiel, 2020-12-05:
I believe you should have downloaded this when installing habitat-sim,
because in the *requirements.txt* file you can find
```attrs>=19.1.0```
I think you can solve this either by:
1- ```pip3 install attrs>=19.1.0```
2- *Which might be better* go to the habitat-sim folder and do: ```pip3 install -r requirements.txt```

KenaHemnani, 2020-12-05:
Thank you for the reply @GhadeerElmkaiel . I have run **pip install -r requirements.txt** inside habitat-sim folder. However, installing this way habitat-sim always gives me **No mule named magnum found error** So I install habitat-sim via following command
**conda install habitat-sim headless -c conda-forge -c aihabitat**  
Hence, habitat-sim and habitat-api are running well after this .
Next, when I run **python main.py -n1 --auto_gpu_config 0 --split val**  inside Neural-SLAM folder 

Following error occured:
(ANS) hemnani1@BOSS8-OS:~/DIR/Neural-SLAM$ python main.py -n1 --auto_gpu_config 0 --split val
Traceback (most recent call last):
  File "main.py", line 15, in <module>
    from env import make_vec_envs
  File "/DATA/hemnani1/DIR/Neural-SLAM/env/__init__.py", line 3, in <module>
    from .habitat import construct_envs
  File "/DATA/hemnani1/DIR/Neural-SLAM/env/habitat/__init__.py", line 9, in <module>
    from .exploration_env import Exploration_Env
  File "/DATA/hemnani1/DIR/Neural-SLAM/env/habitat/exploration_env.py", line 11, in <module>
    from skimage import morphology
ModuleNotFoundError: No module named 'skimage'

Hence, I tried the solution one given in https://github.com/devendrachaplot/Neural-SLAM/issues/1
So, trying with python3.7 

Following error occured:
(ANS1) hemnani1@BOSS8-OS:~/DIR/habitat-api$ python examples/example.py
Traceback (most recent call last):
File "examples/example.py", line 7, in
import habitat
File "/DATA/hemnani1/DIR/habitat-api/habitat/init.py", line 7, in
from habitat.config import Config, get_config
File "/DATA/hemnani1/DIR/habitat-api/habitat/config/init.py", line 7, in
from yacs.config import CfgNode as Config
File "/DATA/hemnani1/.local/lib/python3.7/site-packages/yacs/config.py", line 30, in
import yaml
ModuleNotFoundError: No module named 'yaml'

So, I am stuck here .If you can suggest any solution It will be really helpfull.
Thank you.

GhadeerElmkaiel, 2020-12-05:
it locks for me that you didn't install all the dependencies and requirements for Active-Neural-SLAM.
if ```pip3 install -r requirements.txt``` does not work for some reason just make sure to install everything inside the *requirements.txt* files (**there can be multiple requirements.txt files, one inside the main folder and you might find others in the subfolders**).
for the first error:
```ModuleNotFoundError: No module named 'skimage'``` try to install *scikit-image* using ```pip3 install scikit-image```

for the second error:
```ModuleNotFoundError: No module named 'yaml'``` try to install *yaml* using ```pip install pyyaml```

***Note#1:*** just make sure you are using python3 and activating your (ANS) conda environment when installing packages and python libraries.
***Note#2:*** When using virtual environments and installing packages using pip try not to use ```sudo```, because this will lead to install the packages for the default python, not the environment's python.

KenaHemnani, 2020-12-06:
Hii @GhadeerElmkaiel thank you for response. 
1.) Python3.6.12
Yes, I am installing after activating my virtual environment. when I did **pip3 install scikit-image** It shows me following:
Requirement already satisfied: scikit-image==0.15.0 in /DATA/hemnani1/.local/lib/python3.7/site-packages (from -r requirements.txt (line 6)) (0.15.0)

2.)Python3.7
**Trying pip install pyyaml**  It shows following :
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pyyaml in /usr/local/lib/python3.7/dist-packages (5.3.1)

But here inside /usr/local/lib/python3.7  dist-packages are missing, do you know the reason why????
Hence, packages are already there. But why it is not finding that is weird.

KenaHemnani, 2020-12-06:
Hii @GhadeerElmkaiel  ,
the current status is 

1.) Python3.6:
Installed skicit-image using: **conda install -c conda-forge scikit-image**  solved that particular error. Then also installed scikit-fmm similarly;
So, I got following error:
Traceback (most recent call last):
  File "main.py", line 15, in <module>
    from env import make_vec_envs
  File "/DATA/hemnani1/DIR/Neural-SLAM/env/__init__.py", line 3, in <module>
    from .habitat import construct_envs
  File "/DATA/hemnani1/DIR/Neural-SLAM/env/habitat/__init__.py", line 9, in <module>
    from .exploration_env import Exploration_Env
  File "/DATA/hemnani1/DIR/Neural-SLAM/env/habitat/exploration_env.py", line 29, in <module>
    from env.habitat.utils.noisy_actions import CustomActionSpaceConfiguration
  File "/DATA/hemnani1/DIR/Neural-SLAM/env/habitat/utils/noisy_actions.py", line 12, in <module>
    from habitat_sim.agent.controls import register_move_fn
**ImportError: cannot import name 'register_move_fn'**

2) Python3.7:
After installing pyyaml using conda instead of pip , solved **No module named 'yaml'** .
Now I am encountering following error:

(ANS1) hemnani1@BOSS8-OS:~/DIR/habitat-api$ python examples/example.py                                                                     2020-12-06 15:36:06,378 initializing sim Sim-v0
Traceback (most recent call last):
  File "examples/example.py", line 25, in <module>
    example()
  File "examples/example.py", line 11, in example
    env = habitat.Env(config=habitat.get_config("configs/tasks/pointnav.yaml"))
  File "/DATA/hemnani1/DIR/habitat-api/habitat/core/env.py", line 93, in __init__
    id_sim=self._config.SIMULATOR.TYPE, config=self._config.SIMULATOR
  File "/DATA/hemnani1/DIR/habitat-api/habitat/sims/registration.py", line 19, in                                                           make_sim
    return _sim(**kwargs)
  File "/DATA/hemnani1/DIR/habitat-api/habitat/sims/habitat_simulator/__init__.py                                                          ", line 24, in __init__
    raise habitat_sim_import_error
  File "/DATA/hemnani1/DIR/habitat-api/habitat/sims/habitat_simulator/__init__.py                                                          ", line 7, in _try_register_habitat_sim
    import habitat_sim
  File "/DATA/hemnani1/anaconda3/envs/ANS1/lib/python3.7/site-packages/habitat_si                                                          m/__init__.py", line 13, in <module>
    from habitat_sim import (  # noqa: F401
  File "/DATA/hemnani1/anaconda3/envs/ANS1/lib/python3.7/site-packages/habitat_si                                                          m/agent/__init__.py", line 7, in <module>
    from .agent import *  # noqa: F401,F403
  File "/DATA/hemnani1/anaconda3/envs/ANS1/lib/python3.7/site-packages/habitat_si                                                          m/agent/agent.py", line 10, in <module>
    import magnum as mn
  File "/DATA/hemnani1/.local/lib/python3.7/site-packages/magnum/__init__.py", li                                                          ne 28, in <module>
    from _magnum import *
**ImportError: Python version mismatch: module was compiled for Python 3.6, but the                                                           interpreter version is incompatible: 3.7.9 (default, Aug 31 2020, 12:42:55)**

How, shall I solve this . I have installed everything again in the new environment still saying that module was compiled for Python3.6.

devendrachaplot, 2020-12-06:
Hi,

Some of the above errors seem to be due to different habitat-sim and habitat-api versions. Can you check that you are installing the correct version as specified in the README:
https://github.com/devendrachaplot/Neural-SLAM#installing-dependencies

KenaHemnani, 2020-12-06:
Hii, Sir . Thank you for the response.
Actually, I use  **conda install habitat-sim headless -c conda-forge -c aihabitat** to install habitat-sim as installing via. cloning gives **magnum not found error** always . Is there any way to install this older version of habitat-sim via. conda command? 
Can you please specify that which version numbers are these habitat-sim and habitat-api?

devendrachaplot, 2020-12-07:
We have not tested the code with specific versions of habitat. You can check the exact commit for habitat-sim and habitat-api in the README.

KenaHemnani, 2020-12-11:
Thank you Sir for the response. But, the thing is when I install habitat-sim via. source then neither **python install setup.py** nor **./buld.sh**  works for me. Can you give some suggestions regarding this?  As, via. conda command **conda install habitat-sim headless -c conda-forge -c aihabitat** I cannot check exact commit.

GhadeerElmkaiel, 2020-12-11:
@KenaHemnani can you please show me the error when you build habitat-sim from source.
I might be able to help you. Because I myself have faced multiple problems installing it.😅

KenaHemnani, 2020-12-11:
Hii @GhadeerElmkaiel ,Thank you for response. I mostly get **magnum not found error** installing habitat-sim from source.

KenaHemnani, 2020-12-12:
Hii @GhadeerElmkaiel ;
Here is what I get while installing habitat-sim via. **python setup.py install --headless** :

-- Could NOT find EGL (missing: EGL_INCLUDE_DIR)
CMake Error at deps/magnum/src/Magnum/Platform/CMakeLists.txt:796 (message):
  EGL library, required by some contexts, was not found.  Set
  WITH_*APPLICATION and/or WITH_EGLCONTEXT to OFF to skip building them.

-- Configuring incomplete, errors occurred!
See also "/DATA/hemnani1/DIR/habitat-sim/build/CMakeFiles/CMakeOutput.log".
See also "/DATA/hemnani1/DIR/habitat-sim/build/CMakeFiles/CMakeError.log".
Traceback (most recent call last):
  File "setup.py", line 361, in <module>
    zip_safe=False,
  File "/DATA/hemnani1/anaconda3/envs/ANS/lib/python3.6/site-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/DATA/hemnani1/anaconda3/envs/ANS/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/DATA/hemnani1/anaconda3/envs/ANS/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/DATA/hemnani1/anaconda3/envs/ANS/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/DATA/hemnani1/anaconda3/envs/ANS/lib/python3.6/site-packages/setuptools/command/install.py", line 67, in run
    self.do_egg_install()
  File "/DATA/hemnani1/anaconda3/envs/ANS/lib/python3.6/site-packages/setuptools/command/install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "/DATA/hemnani1/anaconda3/envs/ANS/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/DATA/hemnani1/anaconda3/envs/ANS/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/DATA/hemnani1/anaconda3/envs/ANS/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 167, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/DATA/hemnani1/anaconda3/envs/ANS/lib/python3.6/site-packages/setuptools/command/bdist_egg.py", line 153, in call_command
    self.run_command(cmdname)
  File "/DATA/hemnani1/anaconda3/envs/ANS/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/DATA/hemnani1/anaconda3/envs/ANS/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/DATA/hemnani1/anaconda3/envs/ANS/lib/python3.6/site-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/DATA/hemnani1/anaconda3/envs/ANS/lib/python3.6/distutils/command/install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "/DATA/hemnani1/anaconda3/envs/ANS/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/DATA/hemnani1/anaconda3/envs/ANS/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "setup.py", line 195, in run
    self.build_extension(ext)
  File "setup.py", line 252, in build_extension
    env=env,
  File "/DATA/hemnani1/anaconda3/envs/ANS/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '-H/DATA/hemnani1/DIR/habitat-sim/src', '-Bbuild', '-DBUILD_PYTHON_BINDINGS=ON', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/DATA/hemnani1/DIR/habitat-sim/build/lib.linux-x86_64-3.6/habitat_sim/_ext', '-DPYTHON_EXECUTABLE=/DATA/hemnani1/anaconda3/envs/ANS/bin/python', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-DBUILD_GUI_VIEWERS=OFF', '-DBUILD_TEST=OFF', '-DWITH_BULLET=OFF', '-DBUILD_DATATOOL=OFF']' returned non-zero exit status 1.

You know how to resolve this?

Junior-Zhu, 2020-12-18:
I got the same problem about “Could NOT find EGL” when installing. So I installed ELG LIBRARY via. command **sudo apt-get install libgles2-mesa-dev** and rebooted. This problem was resolved successfully. I think it may works for you.

Skylark0924, 2020-12-19:
> Hi,
> 
> Some of the above errors seem to be due to different habitat-sim and habitat-api versions. Can you check that you are installing the correct version as specified in the README:
> https://github.com/devendrachaplot/Neural-SLAM#installing-dependencies

Hi, I am actually using the same version as you mentioned in the README.md file. However, I still got this error
```
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/home/skylark/.pycharm_helpers/pydev/_pydev_bundle/pydev_umd.py", line 197, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "/home/skylark/.pycharm_helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/home/skylark/Github/Neural-SLAM/main.py", line 15, in <module>
    from env import make_vec_envs
  File "/home/skylark/Github/Neural-SLAM/env/__init__.py", line 3, in <module>
    from .habitat import construct_envs
  File "/home/skylark/Github/Neural-SLAM/env/habitat/__init__.py", line 9, in <module>
    from .exploration_env import Exploration_Env
  File "/home/skylark/Github/Neural-SLAM/env/habitat/exploration_env.py", line 28, in <module>
    from env.habitat.utils.noisy_actions import CustomActionSpaceConfiguration
  File "/home/skylark/Github/Neural-SLAM/env/habitat/utils/noisy_actions.py", line 5, in <module>
    import habitat_sim
  File "/home/lab/anaconda3/envs/habitat_AN/lib/python3.7/site-packages/habitat_sim-0.1.2-py3.7-linux-x86_64.egg/habitat_sim/__init__.py", line 12, in <module>
    from .nav import *
  File "/home/lab/anaconda3/envs/habitat_AN/lib/python3.7/site-packages/habitat_sim-0.1.2-py3.7-linux-x86_64.egg/habitat_sim/nav/__init__.py", line 1, in <module>
    from .greedy_geodesic_follower import GreedyGeodesicFollower
  File "/home/lab/anaconda3/envs/habitat_AN/lib/python3.7/site-packages/habitat_sim-0.1.2-py3.7-linux-x86_64.egg/habitat_sim/nav/greedy_geodesic_follower.py", line 6, in <module>
    import habitat_sim.agent
  File "/home/lab/anaconda3/envs/habitat_AN/lib/python3.7/site-packages/habitat_sim-0.1.2-py3.7-linux-x86_64.egg/habitat_sim/agent/__init__.py", line 7, in <module>
    from .agent import *
  File "/home/lab/anaconda3/envs/habitat_AN/lib/python3.7/site-packages/habitat_sim-0.1.2-py3.7-linux-x86_64.egg/habitat_sim/agent/agent.py", line 10, in <module>
    import magnum as mn
  File "/home/lab/anaconda3/envs/habitat_AN/lib/python3.7/site-packages/magnum/__init__.py", line 28, in <module>
    from _magnum import *
ImportError: Python version mismatch: module was compiled for Python 3.6, but the interpreter version is incompatible: 3.7.9 (default, Aug 31 2020, 12:42:55)
[GCC 7.3.0].
```
As you mentioned in another issue, some problems may occur when using python 3.6 and you recommend using 3.7. But I think this package `magnum` does not fit 3.7.

**ImportError: Python version mismatch: module was compiled for Python 3.6, but the interpreter version is incompatible: 3.7.9**

devendrachaplot, 2021-02-09:
Unfortunately, I can not help much regarding issues with habitat installation. I would suggest installing a fresh copy on anaconda and installing it again.

These are the steps I use to setup the code on a new machine which work for me:

```
# Install conda
wget -O $HOME/miniconda.sh https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh  &&\
    chmod +x ~/miniconda.sh &&\
    ~/miniconda.sh -b -p /custom/conda &&\
    rm ~/miniconda.sh &&\
    /custom/conda/bin/conda install numpy pyyaml scipy ipython mkl mkl-include &&\
    /custom/conda/bin/conda clean -ya
export PATH=/custom/conda/bin:$PATH

# Install cmake
wget https://github.com/Kitware/CMake/releases/download/v3.13.4/cmake-3.13.4-Linux-x86_64.sh
mkdir /custom/cmake
./cmake-3.13.4-Linux-x86_64.sh --prefix=/custom/cmake --skip-license
ln -s /custom/cmake/bin/cmake /usr/local/bin/cmake
cmake --version

# Setup habtiat-sim
git clone https://github.com/facebookresearch/habitat-sim.git
cd habitat-sim; git checkout 9575dcd45fe6f55d2a44043833af08972a7895a9; pip install -r requirements.txt; python setup.py install --headless

# Install challenge specific habitat-api
git clone https://github.com/facebookresearch/habitat-api.git
cd habitat-api; git checkout b5f2b00a25627ecb52b43b13ea96b05998d9a121; pip install -e .
cd habitat-api; wget http://dl.fbaipublicfiles.com/habitat/habitat-test-scenes.zip; unzip habitat-test-scenes.zip

# Test habitat
cd habitat-api; python examples/benchmark.py
```

closed 2021-04-01 by devendrachaplot