devendrachaplot/Neural-SLAM #19: ModuleNotFoundError on runing python main.py -n1 --auto_gpu_config 0 --split val

jsdd25, 2020-12-05:
Hi guys!

I try to test `python main.py -n1 --auto_gpu_config 0 --split val`

I get this error:

(habitat3) jin@jin-MS-7B94:~/Neural-SLAM$ python main.py -n1 --auto_gpu_config 0 --split val
Traceback (most recent call last):
  File "main.py", line 15, in <module>
    from env import make_vec_envs
  File "/home/jin/Neural-SLAM/env/__init__.py", line 3, in <module>
    from .habitat import construct_envs
  File "/home/jin/Neural-SLAM/env/habitat/__init__.py", line 9, in <module>
    from .exploration_env import Exploration_Env
  File "/home/jin/Neural-SLAM/env/habitat/exploration_env.py", line 28, in <module>
    from env.habitat.utils.noisy_actions import CustomActionSpaceConfiguration
  File "/home/jin/Neural-SLAM/env/habitat/utils/noisy_actions.py", line 9, in <module>
    from habitat.sims.habitat_simulator.action_spaces import (
ModuleNotFoundError: No module named 'habitat.sims.habitat_simulator.action_spaces'

Could anyone help me with this? Thank you very much!

devendrachaplot, 2020-12-06:
Hi,

This error is likely due to different habitat-sim and habitat-api versions. Can you check that you are installing the correct version as specified in the README:
https://github.com/devendrachaplot/Neural-SLAM#installing-dependencies

jsdd25, 2020-12-07:
Thank you very much! I have fixed this problem.

closed 2020-12-07 by jsdd25

cky783, 2022-03-04:
I am also meeting the same problem. May I know what specific version works for nerual slam? Thanks a lot!