devendrachaplot/Neural-SLAM #20: Global Policy pretrained model

GhadeerElmkaiel, 2020-12-05:
Hi,
I faced this problem when I tried to use the pretrained model for the Global Policy.
It seems that the size of the used neural network in the code is not the same as the size of the saved model.
I got the following error:
```
Loading global pretrained_models/model_best.global
Traceback (most recent call last):
.
.
.
  File "/media/ubuntu/SSD1/archiconda3/lib/python3.6/site-packages/torch/nn/modules/module.py", line 830, in load_state_dict
    self.__class__.__name__, "\n\t".join(error_msgs)))
RuntimeError: Error(s) in loading state_dict for RL_Policy:
	size mismatch for network.linear1.weight: copying a param with shape torch.Size([256, 7208]) from checkpoint, the shape in current model is torch.Size([256, 31240]).
Exception ignored in: <bound method VectorEnv.__del__ of <env.habitat.habitat_lab.habitat.core.vector_env.VectorEnv object at 0x7efa26acf8>>
```
I downloaded the pretrained models uploaded on [cs.cmu.edu](http://www.cs.cmu.edu/~dchaplot/projects/active_neural_slam/) as follows:
```
wget -O pretrained_models/model_best.global http://www.cs.cmu.edu/~dchaplot/projects/active_neural_slam/model_best.global
```

What is the correct shape of the used Global Policy neural network?

devendrachaplot, 2020-12-06:
Hi,

Does the code run as expected if you do not load the Global Policy model?

Can you give the command you are using to run the code?

GhadeerElmkaiel, 2020-12-07:
Thanks for the fast reply!
I was trying to do some changes to the code to work with MatPlot3d dataset, but I didn't change any thing related to the models, input shapes, output shapes, (I used the same exact models provided  by you).
After some digging in the code I think that I found what can be the cause of the error.
in your [model.py code](https://github.com/devendrachaplot/Neural-SLAM/blob/e833e8f20c5a8d71552a5d22fb78ad6980cfc80c/model.py#L19), when you calculate the out_size of the Global_Policy you use **int()** for the whole expression, while what is happening in the reality that the MaxPool2d is applying **int()** for each axis alone (x-axis and y-axis).
I think that the out_size should be calculated as follows:
```
out_size = int(input_shape[1] / 16.) *int( input_shape[2] / 16.)
``` 
when using the original code with input shape [8, 500, 500] we will get 
```
out_size= int(500/16. * 500/16.) = int(31.25 * 31.25) = int(976,5625) = 976
```
while after applying 4 MaxPool2d(2) we will have an image of size [31, 31], and the out_size will be **out_size=961**

GhadeerElmkaiel, 2020-12-07:
I found the cause of the error, it seems that I have changed the size of the map unintentionally. after fixing that the model can be loaded perfectly.
Sorry It was my fault.

closed 2020-12-07 by GhadeerElmkaiel