devendrachaplot/Neural-SLAM #21: Memory Leak

Anthracenesy, 2020-12-16:
Hi!

We trained the model from scratch on Gibson dataset with the setting that num_scenes = 72, but we found that once the program is running, the avaliable memory of our machine is going lower and lower, and exhausted after about 2 hours. 
We found it leaks over time and stop leaking after we terminate the program.

The curve of avaliable memory is shown as follows.

![image](https://lh3.googleusercontent.com/_NWFCd6TGa9FVv5O5x9brYGutNo6fxSyV5MZSG93jLMd7j_FMeDIw6OlKrD_vK1jeFN-UhVGGUn9xVA1c3cO4hT5VUoRN6rHqVc1s6helNhlD9euxaJFjsyGd8QERqdvchPtWzbLJgnT9XqE-8roXUp2_qC3pVCC4uR9HEg6Dhz5ooNK97BTlleE8sHbcqGxlaj1ZFkKQ1Km9biyA7_U5ZuDc7MrZGvASkaPf6IhyRQe3XGskqokT8MrNTNOlNvKpMGoriFAGAQdCWKOLs6b8e-19_CaHRg00c4pSUWzOkE2eSTWHtW2oHvIf1Dj2Kq3VK7Eoh5IQUIiborvRGMMJaEEj73Il4AwXcP2IIF-U96fwXEUAglY20ZcgiOUHuYURxwX36GVNbS5DP_4bRtQXs_6ZMMLAn0tdnoJUaSagPjO2Om7YH5udjA_e3_7D3iR6HX3-OTFimw8Mw6OcS1RGQcOUDwdNNV5Yz_7PAZdFN8wxVDZurBI8HAIMCJFiYwrZYczywVGEl_L-4DYQSxrXzanzlx6gDrpfEkmVemN8wEF_R9fzCaXh_6_gl0WOvhTtTpuUX9KhvzOsBUWjGGuxLpBFGImC4jMnmWORwf584hi7-Cp4RIj_lszqDacqWoh7Rn7MfqslfAJdFgySE2TjBKC1GxULZGj7ekj3O925BpbcwmS0IW_SK3XCPM=w804-h268-no?authuser=0)

We also found that if we reduce the number of scenes, the speed of memory leaking will also decrease.

It will be very helpful if someone can suggest the possible solution.

Anthracenesy, 2020-12-16:
Hi!

Too long FIFOMemory length may be the cause of Memory Leaking. 

Default value of `slam_memory_size` is set to `500000`, the traning data will be appended to the FIFOMemory untill the FIFO length reaches the maximum of `500000`, which continuously takes up a lot of memory.

We set `slam_memory_size` to `720` hence the queue stop growing and stop consuming the memory.

![](https://lh3.googleusercontent.com/Wy7jmU0qA0ZYkdi_cgufcuJRpjCbPJZLWHvYFyns6U6LU2GayDPs1JEKxP8fOruNyC5ApX_ATEISBrrUj5jCVJx16Xz6bjg_B7WjqRQ57RP6-wYVaKRwBp62qTsSYazwI9J0p-p1y7DNn1VMJ935LgiMr2a6-68zNi4zUNWDfoMWY2AwtPJJd8A6mj5a7KcB4DFRS9v8EueTUq3dlHJUpYBn019mE6npaPrDnzICcAnm4qF5o4VFRoCcIPCxPExCcJLMS02KT3-loEQEAdEu2QxiLNl8RXtPeWhfTnCk67jGnzi75cDpWYzo2KkyYA1HzfPRjlCpYdnnF_k8FsWtxNsPYMSjKMK4_a8fnq-xAjNzVTDt335p49uTeRclrmdnuyW7qKja139v2NbvTe3UeGgjhIJ6dmVIYx4efqhFmPvdnPl6d6Iz_FJi7p_DlpLBH31m9U6HJn-fTkjD3Jtp0cmU_k4CEsfCrqyWHTrIQRoDiw2cgYlIbVlBdwESbprKtajz3cjxjxUCJjrxD8WBZ6xIOnRsLsCv5o3xiEUykPd85ilFnLplrJMXJX3ot-vpXZWHjr915lBWdtZXxEnlUng3DZM-ERzxBKLjf0FQhLNVr8JsQFkZnavUmhDx-fS3lDqKxNUOywYmlLygjLti3ifiZ3xgt90XCsys8S1E_utR14LroxrT9UDwYq8=w805-h266-no?authuser=0)

The large default value of `slam_memory_size` is confusing. Does it matter a lot to the performance of the model?

devendrachaplot, 2021-01-13:
Yes, the memory increase is due to the FIFO memory. In our experiments, we found the memory helps in improving sample efficiency but not the final performance, if you reduce the memory size, the model might require more frames to reach the same performance.

closed 2021-02-09 by devendrachaplot