devendrachaplot/Neural-SLAM #22: Magnum module was compiled for Python 3.6, but the interpreter version is incompatible: 3.7.9

Skylark0924, 2020-12-19:
Hi, I am actually using the same version as you mentioned in the README.md file. However, I still got this error
```
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/home/skylark/.pycharm_helpers/pydev/_pydev_bundle/pydev_umd.py", line 197, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "/home/skylark/.pycharm_helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/home/skylark/Github/Neural-SLAM/main.py", line 15, in <module>
    from env import make_vec_envs
  File "/home/skylark/Github/Neural-SLAM/env/__init__.py", line 3, in <module>
    from .habitat import construct_envs
  File "/home/skylark/Github/Neural-SLAM/env/habitat/__init__.py", line 9, in <module>
    from .exploration_env import Exploration_Env
  File "/home/skylark/Github/Neural-SLAM/env/habitat/exploration_env.py", line 28, in <module>
    from env.habitat.utils.noisy_actions import CustomActionSpaceConfiguration
  File "/home/skylark/Github/Neural-SLAM/env/habitat/utils/noisy_actions.py", line 5, in <module>
    import habitat_sim
  File "/home/lab/anaconda3/envs/habitat_AN/lib/python3.7/site-packages/habitat_sim-0.1.2-py3.7-linux-x86_64.egg/habitat_sim/__init__.py", line 12, in <module>
    from .nav import *
  File "/home/lab/anaconda3/envs/habitat_AN/lib/python3.7/site-packages/habitat_sim-0.1.2-py3.7-linux-x86_64.egg/habitat_sim/nav/__init__.py", line 1, in <module>
    from .greedy_geodesic_follower import GreedyGeodesicFollower
  File "/home/lab/anaconda3/envs/habitat_AN/lib/python3.7/site-packages/habitat_sim-0.1.2-py3.7-linux-x86_64.egg/habitat_sim/nav/greedy_geodesic_follower.py", line 6, in <module>
    import habitat_sim.agent
  File "/home/lab/anaconda3/envs/habitat_AN/lib/python3.7/site-packages/habitat_sim-0.1.2-py3.7-linux-x86_64.egg/habitat_sim/agent/__init__.py", line 7, in <module>
    from .agent import *
  File "/home/lab/anaconda3/envs/habitat_AN/lib/python3.7/site-packages/habitat_sim-0.1.2-py3.7-linux-x86_64.egg/habitat_sim/agent/agent.py", line 10, in <module>
    import magnum as mn
  File "/home/lab/anaconda3/envs/habitat_AN/lib/python3.7/site-packages/magnum/__init__.py", line 28, in <module>
    from _magnum import *
ImportError: Python version mismatch: module was compiled for Python 3.6, but the interpreter version is incompatible: 3.7.9 (default, Aug 31 2020, 12:42:55)
[GCC 7.3.0].
```
As you mentioned in another issue, some problems may occur when using python 3.6 and you recommend using 3.7. But I think this package `magnum` does not fit 3.7.

**ImportError: Python version mismatch: module was compiled for Python 3.6, but the interpreter version is incompatible: 3.7.9**

frankthecr0c, 2021-01-08:
Hi! 
We are able to use both Magnum and python 3.7.9 with anaconda. I think you miss something during the installation procedure. 

Our OS version is Ubuntu 20.04 and Anaconda 4.8.2. 
The procedure we use is the following: 

- Create your Virtual environment with conda with the following parameters and activate the environment: 
                 conda create -n $your_env_name python=3.7 cmake=3.14.0

- Install using pip a pythorch version according to your version of CUDA
- If you use a version of Os that has gcc and g++ >8:

      sudo apt install build-essential 
      sudo apt -y install gcc-7 g++-7 gcc-8 g++-8

      sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-7 7
      sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-7 7
      sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-8 8
      sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-8 8

It is possible to switch between the different versions of gcc and g++ by using the command: update-alternatives.

for C compiler execute:

$ sudo update-alternatives --config gcc

You will be able to choose for the alternative gcc (providing /usr/bin/gcc):

Selection    Path            Priority   Status
------------------------------------------------------------
* 0            /usr/bin/gcc-9   9         auto mode
1            /usr/bin/gcc-7   7         manual mode
2            /usr/bin/gcc-8   8         manual mode
3            /usr/bin/gcc-9   9         manual mode
Press  to keep the current choice[*], or type selection number: 

for C++ compiler execute:

$ sudo update-alternatives --config g++

You will be able to choose for the alternative g++ (providing /usr/bin/gcc):

Selection    Path            Priority   Status
------------------------------------------------------------
* 0            /usr/bin/g++-9   9         auto mode
1            /usr/bin/g++-7   7         manual mode
2            /usr/bin/g++-8   8         manual mode
3            /usr/bin/g++-9   9         manual mode

Press  to keep the current choice[*], or type selection number:

We used gcc and g++ 8
Checking yours active compilers
$ gcc --version
$ g++ --version

Note: rememberr to set the default value (0) after the installation!!
Follow the installation guide to install habitat-sim and habitat-api. 

Note: we also needed to install the following packet to succes (but i think it's not related to your problem, just to inform you): 
libjpeg-dev libglm-dev libgl1-mesa-glx libegl1-mesa-dev mesa-utils xorg-dev freeglut3-dev libbullet-dev

Skylark0924, 2021-01-08:
Thank you for your reply! I will test it later.

devendrachaplot, 2021-02-17:
Closing due to inactivity.

closed 2021-02-17 by devendrachaplot