devendrachaplot/Neural-SLAM #27: Running paper experiments + running Matterport based experiments

shurjobanerjee, 2021-01-22:
Is there a simple set of instructions available for running and reproducing experiments from the paper? I can't seem to find the relevant documentation anywhere.

Also, are there a set of concrete exact steps for running the code on Matterport? I have tried following along the steps provided in #16 but was hoping there was a better way?
 
 Thanks!

devendrachaplot, 2021-02-09:
Training and evaluation instructions are available here:
https://github.com/devendrachaplot/Neural-SLAM#usage

The pretrained models should give you numbers slightly above those reported in the paper.

Regarding Matterport, can you check if the Matterport dataset is working with the habitat-api/habitat-lab (without using the Neural-SLAM code)?

Junior-Zhu, 2021-03-29:
> Is there a simple set of instructions available for running and reproducing experiments from the paper? I can't seem to find the relevant documentation anywhere.
> 
> Also, are there a set of concrete exact steps for running the code on Matterport? I have tried following along the steps provided in #16 but was hoping there was a better way?
> Simply using the mp3d configuration file leads to a `Trade::AbstractImporter::image2D(): index out of range` error.
> 
> Thanks!

Hi,
I also got the same error when runing with the mp3d dataset. Did you solve it?

shurjobanerjee, 2021-04-12:
I did! I switched the habitat_api folder inside env/habitat (https://github.com/devendrachaplot/Neural-SLAM/tree/master/env/habitat)  to the latest version of habitat.
I than transferred the changes from this commit: https://github.com/devendrachaplot/habitat-api/commit/8ac1b16aa0554b10748a925f8a937954c77c1563#diff-c0fd02c963c59f63d0125bf919c5fd5cbfeb0c20d8a12a8f7a2a17a5122f1de8 to the new habitat environment.

closed 2021-04-12 by shurjobanerjee