devendrachaplot/Neural-SLAM #34: Broken pipe caused by ZeroDivisionError

DEQDON, 2021-03-23:
Hi, I ran into a ZeroDivisionError which causes Broken pipe. The command I ran was `python main.py -n1 --auto_gpu_config 0 --split val
`
At first, the code was all good, and I could see the training process with losses and stuff. But after some time, it broke down and the error log is as follows:

```
2021-03-24 00:41:29,566 Computing map for data/scene_datasets/gibson/Cantwell.glb
Traceback (most recent call last):
  File "main.py", line 769, in <module>
    main()
  File "main.py", line 617, in main
    g_agent.update(g_rollouts)
  File "/home/xxx/Neural-SLAM/algo/ppo.py", line 58, in update
    for sample in data_generator:
  File "/home/xxx/Neural-SLAM/utils/storage.py", line 95, in feed_forward_generator
    mini_batch_size = batch_size // num_mini_batch
ZeroDivisionError: integer division or modulo by zero
Exception ignored in: <function VectorEnv.__del__ at 0x7ff66e4a1e60>
Traceback (most recent call last):
  File "/home/xxx/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 487, in __del__
    self.close()
  File "/home/xxx/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 351, in close
    write_fn((CLOSE_COMMAND, None))
  File "/home/xxx/anaconda3/envs/comp765/lib/python3.7/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/home/xxx/anaconda3/envs/comp765/lib/python3.7/multiprocessing/connection.py", line 404, in _send_bytes
    self._send(header + buf)
  File "/home/xxx/anaconda3/envs/comp765/lib/python3.7/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
BrokenPipeError: [Errno 32] Broken pipe
```

Could anyone tell me what to do with this?

SJingwen, 2021-03-29:
I met the same error. Have you solved it?

devendrachaplot, 2021-04-01:
Please add `--num_mini_batch 1` to the command, this should fix the error.

DEQDON, 2021-04-10:
Thanks, this works!

closed 2021-04-10 by DEQDON