devendrachaplot/Neural-SLAM #41: ImportError: /usr/local/lib/python3.6/dist-packages/_magnum.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _Py_ZeroStruct

emontero1991, 2021-07-13:
I'm trying to implement Active Neural SLAM by following the Github post. But has the following error. Please help me if any solution.....

I'm using ubuntu 18.04, python 3.6.9
Got this error when checking the dependencies   ----- sudo  python3 main.py -n1 --auto_gpu_config 0 --split val

Traceback (most recent call last):
  File "main.py", line 15, in <module>
    from env import make_vec_envs
  File "/home/user/Neural-SLAM/env/__init__.py", line 3, in <module>
    from .habitat import construct_envs
  File "/home/user/Neural-SLAM/env/habitat/__init__.py", line 9, in <module>
    from .exploration_env import Exploration_Env
  File "/home/user/Neural-SLAM/env/habitat/exploration_env.py", line 28, in <module>
    from env.habitat.utils.noisy_actions import CustomActionSpaceConfiguration
  File "/home/user/Neural-SLAM/env/habitat/utils/noisy_actions.py", line 5, in <module>
    import habitat_sim
  File "/usr/local/lib/python3.6/dist-packages/habitat_sim-0.1.2-py3.6-linux-x86_64.egg/habitat_sim/__init__.py", line 12, in <module>
    from .nav import *
  File "/usr/local/lib/python3.6/dist-packages/habitat_sim-0.1.2-py3.6-linux-x86_64.egg/habitat_sim/nav/__init__.py", line 1, in <module>
    from .greedy_geodesic_follower import GreedyGeodesicFollower
  File "/usr/local/lib/python3.6/dist-packages/habitat_sim-0.1.2-py3.6-linux-x86_64.egg/habitat_sim/nav/greedy_geodesic_follower.py", line 6, in <module>
    import habitat_sim.agent
  File "/usr/local/lib/python3.6/dist-packages/habitat_sim-0.1.2-py3.6-linux-x86_64.egg/habitat_sim/agent/__init__.py", line 7, in <module>
    from .agent import *
  File "/usr/local/lib/python3.6/dist-packages/habitat_sim-0.1.2-py3.6-linux-x86_64.egg/habitat_sim/agent/agent.py", line 10, in <module>
    import magnum as mn
  File "/usr/local/lib/python3.6/dist-packages/magnum/__init__.py", line 28, in <module>
    from _magnum import *
ImportError: /usr/local/lib/python3.6/dist-packages/_magnum.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _Py_ZeroStruct

devendrachaplot, 2021-07-19:
This seems to be an issue with habitat installation. Quick way to check this is by running examples/benchmark.py in habitat-api directory (where you installed habitat-api, not the submodule within Neural-SLAM directory). If it throws an error, it indicates habitat-sim or api is not installed correctly.

emontero1991, 2021-07-20:
Thank you so much for the response.
We got succeeded in the installation of Habitat-sim and Active Neural SLAM.

By using a virtual python environment problem got resolved. (venv). 

Earlier wrong version of python was used for compiling. 
When using setup.py, we need to run that command outside the root repo folder, otherwise, you'll pick up the source instead of what was installed, possibly causing conflicts.

emontero1991, 2021-07-20:
sorry to bother you again.
We have another issue.
#################################################################
**System Specifications :**
Ubuntu 20.04
Habitat-sim version  0.1.2
Python version 3.8.10
GPU 1 (Nvidia Titan V - 12 gb)
CPU -  12 core 24 thread processor
###########################################################
This error arises for both training and evaluation.
Evaluation comment: 
python main.py --split val --eval 1 --train_global 0 --train_local 0 --train_slam 0 \
--load_global pretrained_models/model_best.global \
--load_local pretrained_models/model_best.local \
--load_slam pretrained_models/model_best.slam 

Auto GPU config:
gpu_memory: 11.78082275390625
Number of processes: 1
Number of processes on GPU 0: 1
Number of processes per GPU: 0
Dumping at ./tmp//models/exp1/
Namespace(alpha=0.99, auto_gpu_config=1, camera_height=1.25, clip_param=0.2, collision_threshold=0.2, cuda=True, du_scale=2, dump_location='./tmp/', entropy_coef=0.001, env_frame_height=256, env_frame_width=256, eps=1e-05, eval=1, exp_loss_coeff=1.0, exp_name='exp1', frame_height=128, frame_width=128, gamma=0.99, global_downscaling=2, global_hidden_size=256, global_lr=2.5e-05, goals_size=2, hfov=90.0, load_global='pretrained_models/model_best.global', load_local='pretrained_models/model_best.local', load_slam='pretrained_models/model_best.slam', local_hidden_size=512, local_optimizer='adam,lr=0.0001', local_policy_update_freq=5, log_interval=10, map_pred_threshold=0.5, map_resolution=5, map_size_cm=2400, max_episode_length=1000, max_grad_norm=0.5, no_cuda=False, noise_level=1.0, noisy_actions=1, noisy_odometry=1, num_episodes=1000000, num_global_steps=40, num_local_steps=25, num_mini_batch=0, num_processes=1, num_processes_on_first_gpu=1, num_processes_per_gpu=0, obs_threshold=1, obstacle_boundary=5, pose_loss_coeff=10000.0, ppo_epoch=4, pretrained_resnet=1, print_images=0, proj_loss_coeff=1.0, randomize_env_every=1000, save_interval=1, save_periodic=500000, save_trajectory_data='0', seed=1, short_goal_dist=1, sim_gpu_id=1, slam_batch_size=72, slam_iterations=10, slam_memory_size=1000000, slam_optimizer='adam,lr=0.0001', split='val', task_config='tasks/pointnav_gibson.yaml', tau=0.95, total_num_scenes=1, train_global=0, train_local=0, train_slam=0, use_deterministic_local=0, use_gae=False, use_pose_estimation=2, use_recurrent_global=0, use_recurrent_local=1, value_loss_coef=0.5, vis_type=1, vision_range=64, visualize=1)
Auto GPU config:
gpu_memory: 11.78082275390625
Number of processes: 1
Number of processes on GPU 0: 1
Number of processes per GPU: 0
Loading data/scene_datasets/gibson/Cantwell.glb
2021-07-20 13:47:58,357 initializing sim Sim-v0
Renderer: NVIDIA TITAN V/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 470.42.01
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
I0720 13:48:00.368867 24203 simulator.py:80] Loaded navmesh data/scene_datasets/gibson/Cantwell.navmesh
2021-07-20 13:48:00,369 initializing task Nav-v0
2021-07-20 13:48:00,375 Computing map for data/scene_datasets/gibson/Cantwell.glb
/home/msdl/my-project-env/lib/python3.8/site-packages/torch/nn/functional.py:3289: UserWarning: Default grid_sample and affine_grid behavior has changed to align_corners=False since 1.3.0. Please specify align_corners=True if the old behavior is desired. See the documentation of grid_sample for details.
  warnings.warn("Default grid_sample and affine_grid behavior has changed "
/home/msdl/my-project-env/lib/python3.8/site-packages/torch/nn/functional.py:3226: UserWarning: Default grid_sample and affine_grid behavior has changed to align_corners=False since 1.3.0. Please specify align_corners=True if the old behavior is desired. See the documentation of grid_sample for details.
  warnings.warn("Default grid_sample and affine_grid behavior has changed "
Loading slam pretrained_models/model_best.slam
Traceback (most recent call last):
  File "main.py", line 769, in <module>
    main()
  File "main.py", line 240, in main
    state_dict = torch.load(args.load_slam,
  File "/home/msdl/my-project-env/lib/python3.8/site-packages/torch/serialization.py", line 593, in load
    return _legacy_load(opened_file, map_location, pickle_module, **pickle_load_args)
  File "/home/msdl/my-project-env/lib/python3.8/site-packages/torch/serialization.py", line 763, in _legacy_load
    magic_number = pickle_module.load(f, **pickle_load_args)
_pickle.UnpicklingError: invalid load key, '<'.
Exception ignored in: <function VectorEnv.__del__ at 0x7fad69275b80>
Traceback (most recent call last):
  File "/home/msdl/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 487, in __del__
    self.close()
  File "/home/msdl/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 351, in close
    write_fn((CLOSE_COMMAND, None))
[... cut ...] send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/usr/lib/python3.8/multiprocessing/connection.py", line 411, in _send_bytes
    self._send(header + buf)
  File "/usr/lib/python3.8/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
BrokenPipeError: [Errno 32] Broken pipe

Is this error because of pickle or memory?
We have changed the slam memory size ('-sms', '--slam_memory_size') but did not succeed.
Looking forward for your reply.
Thank you so much

devendrachaplot, 2022-09-09:
This looks like a pickle error and seems unrelated to habitat-sim.

closed 2022-11-07 by devendrachaplot