devendrachaplot/Neural-SLAM #47: Frontier-based exploration

AgentEXPL, 2021-09-03:
Hi, @devendrachaplot . Thanks for sharing this great work. From the paper, I know that Frontier-based exploration was also implemented. I'm sorry that I did not find this part in the provided code. It would be of great help if I could be provided with the code of Frontier-based exploration.

marcopremigit, 2022-02-04:
> Hi, @devendrachaplot . Thanks for sharing this great work. From the paper, I know that Frontier-based exploration was also implemented. I'm sorry that I did not find this part in the provided code. It would be of great help if I could be provided with the code of Frontier-based exploration.

Hi, did you find a way to test the code with Frontier-based exploration?

AgentEXPL, 2022-02-08:
Hi @marcopremigit , the answer is no.