devendrachaplot/Neural-SLAM #58: two question about GPU and visualization

GauleeJX, 2022-07-12:
I'm so sorry about questioning!
1. when i use multiple gpus(for example arguments are -n 10 --num_processes_per_gpu 5) i will get the following error:  
  WindowlessContext.cpp:227] check failed: device == 0 (1 vs. 0) glX context does not support multiple GPUs. Please compile with BUILD_GUI_VIEWERS=0 for multi-gpu support via EGL
And There are 4 3090 GPUs  on my computer.

2. i add -v 1 as an argument but nothing happen, there are no observations and errors about something. 

How can i fix these?
I really apprecita any help you can provide.

devendrachaplot, 2022-09-09:
Could you try running this with '-n 1 -v 1'?

fatPeter, 2022-10-22:
```python
if sys.platform == 'darwin':
    matplotlib.use("tkagg")
else:
    matplotlib.use('Agg')
```
For visualization, simply comment these lines. 
It works for my linux.

closed 2022-11-07 by devendrachaplot