devendrachaplot/Neural-SLAM #60: single GPU error

frischzenger, 2022-09-09:
python main.py -n1 --auto_gpu_config 0 --split val
----------------------------------------------------------------------------------------------
Dumping at ./tmp//models/exp1/
Namespace(alpha=0.99, auto_gpu_config=0, camera_height=1.25, clip_param=0.2, collision_threshold=0.2, cuda=True, du_scale=2, dump_location='./tmp/', entropy_coef=0.001, env_frame_height=256, env_frame_width=256, eps=1e-05, eval=0, exp_loss_coeff=1.0, exp_name='exp1', frame_height=128, frame_width=128, gamma=0.99, global_downscaling=2, global_hidden_size=256, global_lr=2.5e-05, goals_size=2, hfov=90.0, load_global='0', load_local='0', load_slam='0', local_hidden_size=512, local_optimizer='adam,lr=0.0001', local_policy_update_freq=5, log_interval=10, map_pred_threshold=0.5, map_resolution=5, map_size_cm=2400, max_episode_length=1000, max_grad_norm=0.5, no_cuda=False, noise_level=1.0, noisy_actions=1, noisy_odometry=1, num_episodes=1000000, num_global_steps=40, num_local_steps=25, num_mini_batch=0, num_processes=1, num_processes_on_first_gpu=0, num_processes_per_gpu=11, obs_threshold=1, obstacle_boundary=5, pose_loss_coeff=10000.0, ppo_epoch=4, pretrained_resnet=1, print_images=0, proj_loss_coeff=1.0, randomize_env_every=1000, save_interval=1, save_periodic=500000, save_trajectory_data='0', seed=1, short_goal_dist=1, sim_gpu_id=0, slam_batch_size=12, slam_iterations=10, slam_memory_size=16000, slam_optimizer='adam,lr=0.0001', split='val', task_config='tasks/pointnav_gibson.yaml', tau=0.95, total_num_scenes='auto', train_global=1, train_local=1, train_slam=1, use_deterministic_local=0, use_gae=False, use_pose_estimation=2, use_recurrent_global=0, use_recurrent_local=1, value_loss_coef=0.5, vis_type=1, vision_range=64, visualize=0)
Loading data/scene_datasets/gibson/Cantwell.glb
2022-09-09 14:49:32,279 initializing sim Sim-v0
Renderer: NVIDIA GeForce GTX 1080/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 515.65.01
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
I0909 14:49:35.622740 336 simulator.py:80] Loaded navmesh data/scene_datasets/gibson/Cantwell.navmesh
2022-09-09 14:49:35,624 initializing task Nav-v0
Traceback (most recent call last):
  File "main.py", line 776, in <module>
    main()
  File "main.py", line 119, in main
    envs = make_vec_envs(args)
  File "/home/frisch/project/Neural-SLAM/env/__init__.py", line 9, in make_vec_envs
    envs = VecPyTorch(envs, args.device)
  File "/home/frisch/project/Neural-SLAM/env/__init__.py", line 19, in __init__
    self.observation_space = venv.observation_space
AttributeError: 'VectorEnv' object has no attribute 'observation_space'
Exception ignored in: <function VectorEnv.__del__ at 0x7f0e24b4a950>
Traceback (most recent call last):
  File "/home/frisch/project/habitat-api/habitat/core/vector_env.py", line 458, in __del__
    self.close()
  File "/home/frisch/project/habitat-api/habitat/core/vector_env.py", line 336, in close
    write_fn((CLOSE_COMMAND, None))
  File "/home/frisch/anaconda3/envs/habitat/lib/python3.7/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/home/frisch/anaconda3/envs/habitat/lib/python3.7/multiprocessing/connection.py", line 404, in _send_bytes
    self._send(header + buf)
  File "/home/frisch/anaconda3/envs/habitat/lib/python3.7/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
BrokenPipeError: [Errno 32] Broken pipe
(habitat) frisch@frisch:~/project/Neural-SLAM$ python
Python 3.7.13 (default, Mar 29 2022, 02:18:16) 
[GCC 7.5.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gym
>>> gym.__version__
'0.10.9'
>>> exit()
(habitat) frisch@frisch:~/project/Neural-SLAM$ python main.py -n1 --auto_gpu_config 0 --split val
Dumping at ./tmp//models/exp1/
Namespace(alpha=0.99, auto_gpu_config=0, camera_height=1.25, clip_param=0.2, collision_threshold=0.2, cuda=True, du_scale=2, dump_location='./tmp/', entropy_coef=0.001, env_frame_height=256, env_frame_width=256, eps=1e-05, eval=0, exp_loss_coeff=1.0, exp_name='exp1', frame_height=128, frame_width=128, gamma=0.99, global_downscaling=2, global_hidden_size=256, global_lr=2.5e-05, goals_size=2, hfov=90.0, load_global='0', load_local='0', load_slam='0', local_hidden_size=512, local_optimizer='adam,lr=0.0001', local_policy_update_freq=5, log_interval=10, map_pred_threshold=0.5, map_resolution=5, map_size_cm=2400, max_episode_length=1000, max_grad_norm=0.5, no_cuda=False, noise_level=1.0, noisy_actions=1, noisy_odometry=1, num_episodes=1000000, num_global_steps=40, num_local_steps=25, num_mini_batch=0, num_processes=1, num_processes_on_first_gpu=0, num_processes_per_gpu=5, obs_threshold=1, obstacle_boundary=5, pose_loss_coeff=10000.0, ppo_epoch=4, pretrained_resnet=1, print_images=0, proj_loss_coeff=1.0, randomize_env_every=1000, save_interval=1, save_periodic=500000, save_trajectory_data='0', seed=1, short_goal_dist=1, sim_gpu_id=0, slam_batch_size=12, slam_iterations=10, slam_memory_size=16000, slam_optimizer='adam,lr=0.0001', split='val', task_config='tasks/pointnav_gibson.yaml', tau=0.95, total_num_scenes='auto', train_global=1, train_local=1, train_slam=1, use_deterministic_local=0, use_gae=False, use_pose_estimation=2, use_recurrent_global=0, use_recurrent_local=1, value_loss_coef=0.5, vis_type=1, vision_range=64, visualize=0)
Loading data/scene_datasets/gibson/Cantwell.glb
2022-09-09 15:17:55,843 initializing sim Sim-v0
Renderer: NVIDIA GeForce GTX 1080/PCIe/SSE2 by NVIDIA Corporation
OpenGL version: 4.6.0 NVIDIA 515.65.01
Using optional features:
    GL_ARB_ES2_compatibility
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_invalidate_subdata
    GL_ARB_multi_bind
    GL_ARB_robustness
    GL_ARB_separate_shader_objects
    GL_ARB_texture_filter_anisotropic
    GL_ARB_texture_storage
    GL_ARB_texture_storage_multisample
    GL_ARB_vertex_array_object
    GL_KHR_debug
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
    nv-zero-context-profile-mask
    nv-implementation-color-read-format-dsa-broken
    nv-cubemap-inconsistent-compressed-image-size
    nv-cubemap-broken-full-compressed-image-query
    nv-compressed-block-size-in-bits
I0909 15:17:59.113152 1093 simulator.py:80] Loaded navmesh data/scene_datasets/gibson/Cantwell.navmesh
2022-09-09 15:17:59,113 initializing task Nav-v0
Traceback (most recent call last):
  File "main.py", line 776, in <module>
    main()
  File "main.py", line 119, in main
    envs = make_vec_envs(args)
  File "/home/frisch/project/Neural-SLAM/env/__init__.py", line 9, in make_vec_envs
    envs = VecPyTorch(envs, args.device)
  File "/home/frisch/project/Neural-SLAM/env/__init__.py", line 20, in __init__
    self.observation_space = venv.observation_space
AttributeError: 'VectorEnv' object has no attribute 'observation_space'
Exception ignored in: <function VectorEnv.__del__ at 0x7fafd9db3950>
Traceback (most recent call last):
  File "/home/frisch/project/habitat-api/habitat/core/vector_env.py", line 458, in __del__
    self.close()
  File "/home/frisch/project/habitat-api/habitat/core/vector_env.py", line 336, in close
    write_fn((CLOSE_COMMAND, None))
  File "/home/frisch/anaconda3/envs/habitat/lib/python3.7/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/home/frisch/anaconda3/envs/habitat/lib/python3.7/multiprocessing/connection.py", line 404, in _send_bytes
    self._send(header + buf)
  File "/home/frisch/anaconda3/envs/habitat/lib/python3.7/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
BrokenPipeError: [Errno 32] Broken pipe

frischzenger, 2022-09-09:
it report : AttributeError: 'VectorEnv' object has no attribute 'observation_space'

devendrachaplot, 2022-09-09:
Could you verify that you have installed the correct older versions of habitat-sim and habitat-api as described in the README?

closed 2022-11-07 by devendrachaplot

Ran12256, 2025-03-27:
> it report : AttributeError: 'VectorEnv' object has no attribute 'observation_space'

hi, have you solved the problem? I met the same problem. Thank you!

Ran12256, 2025-03-27:
> ## python main.py -n1 --auto_gpu_config 0 --split val
> Dumping at ./tmp//models/exp1/ Namespace(alpha=0.99, auto_gpu_config=0, camera_height=1.25, clip_param=0.2, collision_threshold=0.2, cuda=True, du_scale=2, dump_location='./tmp/', entropy_coef=0.001, env_frame_height=256, env_frame_width=256, eps=1e-05, eval=0, exp_loss_coeff=1.0, exp_name='exp1', frame_height=128, frame_width=128, gamma=0.99, global_downscaling=2, global_hidden_size=256, global_lr=2.5e-05, goals_size=2, hfov=90.0, load_global='0', load_local='0', load_slam='0', local_hidden_size=512, local_optimizer='adam,lr=0.0001', local_policy_update_freq=5, log_interval=10, map_pred_threshold=0.5, map_resolution=5, map_size_cm=2400, max_episode_length=1000, max_grad_norm=0.5, no_cuda=False, noise_level=1.0, noisy_actions=1, noisy_odometry=1, num_episodes=1000000, num_global_steps=40, num_local_steps=25, num_mini_batch=0, num_processes=1, num_processes_on_first_gpu=0, num_processes_per_gpu=11, obs_threshold=1, obstacle_boundary=5, pose_loss_coeff=10000.0, ppo_epoch=4, pretrained_resnet=1, print_images=0, proj_loss_coeff=1.0, randomize_env_every=1000, save_interval=1, save_periodic=500000, save_trajectory_data='0', seed=1, short_goal_dist=1, sim_gpu_id=0, slam_batch_size=12, slam_iterations=10, slam_memory_size=16000, slam_optimizer='adam,lr=0.0001', split='val', task_config='tasks/pointnav_gibson.yaml', tau=0.95, total_num_scenes='auto', train_global=1, train_local=1, train_slam=1, use_deterministic_local=0, use_gae=False, use_pose_estimation=2, use_recurrent_global=0, use_recurrent_local=1, value_loss_coef=0.5, vis_type=1, vision_range=64, visualize=0) Loading data/scene_datasets/gibson/Cantwell.glb 2022-09-09 14:49:32,279 initializing sim Sim-v0 Renderer: NVIDIA GeForce GTX 1080/PCIe/SSE2 by NVIDIA Corporation OpenGL version: 4.6.0 NVIDIA 515.65.01 Using optional features: GL_ARB_ES2_compatibility GL_ARB_direct_state_access GL_ARB_get_texture_sub_image GL_ARB_invalidate_subdata GL_ARB_multi_bind GL_ARB_robustness GL_ARB_separate_shader_objects GL_ARB_texture_filter_anisotropic GL_ARB_texture_storage GL_ARB_texture_storage_multisample GL_ARB_vertex_array_object GL_KHR_debug Using driver workarounds: no-layout-qualifiers-on-old-glsl nv-zero-context-profile-mask nv-implementation-color-read-format-dsa-broken nv-cubemap-inconsistent-compressed-image-size nv-cubemap-broken-full-compressed-image-query nv-compressed-block-size-in-bits I0909 14:49:35.622740 336 simulator.py:80] Loaded navmesh data/scene_datasets/gibson/Cantwell.navmesh 2022-09-09 14:49:35,624 initializing task Nav-v0 Traceback (most recent call last): File "main.py", line 776, in main() File "main.py", line 119, in main envs = make_vec_envs(args) File "/home/frisch/project/Neural-SLAM/env/**init**.py", line 9, in make_vec_envs envs = VecPyTorch(envs, args.device) File "/home/frisch/project/Neural-SLAM/env/**init**.py", line 19, in **init** self.observation_space = venv.observation_space AttributeError: 'VectorEnv' object has no attribute 'observation_space' Exception ignored in: <function VectorEnv.**del** at 0x7f0e24b4a950> Traceback (most recent call last): File "/home/frisch/project/habitat-api/habitat/core/vector_env.py", line 458, in **del** self.close() File "/home/frisch/project/habitat-api/habitat/core/vector_env.py", line 336, in close write_fn((CLOSE_COMMAND, None)) File "/home/frisch/anaconda3/envs/habitat/lib/python3.7/multiprocessing/connection.py", line 206, in send self._send_bytes(_ForkingPickler.dumps(obj)) File "/home/frisch/anaconda3/envs/habitat/lib/python3.7/multiprocessing/connection.py", line 404, in _send_bytes self._send(header + buf) File "/home/frisch/anaconda3/envs/habitat/lib/python3.7/multiprocessing/connection.py", line 368, in _send n = write(self._handle, buf) BrokenPipeError: [Errno 32] Broken pipe (habitat) frisch@frisch:~/project/Neural-SLAM$ python Python 3.7.13 (default, Mar 29 2022, 02:18:16) [GCC 7.5.0] :: Anaconda, Inc. on linux Type "help", "copyright", "credits" or "license" for more information.
> 
> > > > import gym
> > > > gym.**version**
> > > > '0.10.9'
> > > > exit()
> > > > (habitat) frisch@frisch:~/project/Neural-SLAM$ python main.py -n1 --auto_gpu_config 0 --split val
> > > > Dumping at ./tmp//models/exp1/
> > > > Namespace(alpha=0.99, auto_gpu_config=0, camera_height=1.25, clip_param=0.2, collision_threshold=0.2, cuda=True, du_scale=2, dump_location='./tmp/', entropy_coef=0.001, env_frame_height=256, env_frame_width=256, eps=1e-05, eval=0, exp_loss_coeff=1.0, exp_name='exp1', frame_height=128, frame_width=128, gamma=0.99, global_downscaling=2, global_hidden_size=256, global_lr=2.5e-05, goals_size=2, hfov=90.0, load_global='0', load_local='0', load_slam='0', local_hidden_size=512, local_optimizer='adam,lr=0.0001', local_policy_update_freq=5, log_interval=10, map_pred_threshold=0.5, map_resolution=5, map_size_cm=2400, max_episode_length=1000, max_grad_norm=0.5, no_cuda=False, noise_level=1.0, noisy_actions=1, noisy_odometry=1, num_episodes=1000000, num_global_steps=40, num_local_steps=25, num_mini_batch=0, num_processes=1, num_processes_on_first_gpu=0, num_processes_per_gpu=5, obs_threshold=1, obstacle_boundary=5, pose_loss_coeff=10000.0, ppo_epoch=4, pretrained_resnet=1, print_images=0, proj_loss_coeff=1.0, randomize_env_every=1000, save_interval=1, save_periodic=500000, save_trajectory_data='0', seed=1, short_goal_dist=1, sim_gpu_id=0, slam_batch_size=12, slam_iterations=10, slam_memory_size=16000, slam_optimizer='adam,lr=0 [... cut ...]
> > > > File "/home/frisch/anaconda3/envs/habitat/lib/python3.7/multiprocessing/connection.py", line 206, in send
> > > > self._send_bytes(_ForkingPickler.dumps(obj))
> > > > File "/home/frisch/anaconda3/envs/habitat/lib/python3.7/multiprocessing/connection.py", line 404, in _send_bytes
> > > > self._send(header + buf)
> > > > File "/home/frisch/anaconda3/envs/habitat/lib/python3.7/multiprocessing/connection.py", line 368, in _send
> > > > n = write(self._handle, buf)
> > > > BrokenPipeError: [Errno 32] Broken pipe

hi, have you solved the problem? I met the same problem. Thank you!

frischzenger, 2025-05-13:
> hi, have you solved the problem? I met the same problem. Thank you!

use old habitat-sim and habitat-api