devendrachaplot/Neural-SLAM #64: ModuleNotFoundError: No module named 'habitat'

rushibs, 2022-12-28:
I followed all the steps mentioned to reproduce the codebase. However, I am getting the error shown below when I run the following command: 

python3 main.py -n1 --auto_gpu_config 0 --split val

ERROR:

 Traceback (most recent call last):
  File "main.py", line 15, in <module>
    from env import make_vec_envs
  File "/home/USER/Neural-SLAM/env/__init__.py", line 3, in <module>
    from .habitat import construct_envs
  File "/home/USER/Neural-SLAM/env/habitat/__init__.py", line 6, in <module>
    from habitat.config.default import get_config as cfg_env
ModuleNotFoundError: No module named 'habitat'

Could you kindly guide me what the issue could be and how to resolve it?

snowBrain, 2023-10-13:
# import env.habitat.utils.pose as pu
from env.habitat.utils import pose as pu
#import env.habitat.utils.visualizations as vu
from env.habitat.utils import visualizations as vu

sahusiddharth, 2024-04-18:
Hi @rushibs, were you able to resolve it?

Hi @snowBrain, I tried what you suggested but the error is still there, can you please help me resolve it?