devendrachaplot/Neural-SLAM #70: BrokenPipeError: [Errno 32] Broken pipe

daiki240, 2024-05-29:
Hi,
After configuring according to readme.md, I run the command python main.py -n1 --auto_gpu_config 0 --split val and get the following error.

Traceback (most recent call last):
  File "main.py", line 770, in <module>
    main()
  File "main.py", line 120, in main
    envs = make_vec_envs(args)
  File "/share_dir/add_hdd/iwata/Neural-SLAM/env/__init__.py", line 7, in make_vec_envs
    envs = construct_envs(args)
  File "/share_dir/add_hdd/iwata/Neural-SLAM/env/habitat/__init__.py", line 102, in construct_envs
    range(args.num_processes))
  File "/share_dir/add_hdd/iwata/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 117, in __init__
    read_fn() for read_fn in self._connection_read_fns
  File "/share_dir/add_hdd/iwata/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 117, in <listcomp>
    read_fn() for read_fn in self._connection_read_fns
  File "/opt/conda/envs/v1/lib/python3.7/multiprocessing/connection.py", line 250, in recv
    buf = self._recv_bytes()
  File "/opt/conda/envs/v1/lib/python3.7/multiprocessing/connection.py", line 407, in _recv_bytes
    buf = self._recv(4)
  File "/opt/conda/envs/v1/lib/python3.7/multiprocessing/connection.py", line 379, in _recv
    chunk = read(handle, remaining)
ConnectionResetError: [Errno 104] Connection reset by peer
Exception ignored in: <function VectorEnv.__del__ at 0x7f873050d950>
Traceback (most recent call last):
  File "/share_dir/add_hdd/iwata/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 487, in __del__
    self.close()
  File "/share_dir/add_hdd/iwata/Neural-SLAM/env/habitat/habitat_api/habitat/core/vector_env.py", line 351, in close
    write_fn((CLOSE_COMMAND, None))
  File "/opt/conda/envs/v1/lib/python3.7/multiprocessing/connection.py", line 206, in send
    self._send_bytes(_ForkingPickler.dumps(obj))
  File "/opt/conda/envs/v1/lib/python3.7/multiprocessing/connection.py", line 404, in _send_bytes
    self._send(header + buf)
  File "/opt/conda/envs/v1/lib/python3.7/multiprocessing/connection.py", line 368, in _send
    n = write(self._handle, buf)
BrokenPipeError: [Errno 32] Broken pipe

Datasets are arranged as described in readme.md.

linaug, 2024-06-05:
Hi, I have other questions about install env. And Do you known how to download the data folder, which is mentioned in the Readme. I cann't find the scene_datasets and datasets in the website (https:/(github.com/facebookresearch/habitat-api#data).

daiki240, 2024-06-06:
I have not been able to get the code to work correctly yet, so I don't know if it is correct,
https://github.com/facebookresearch/habitat-lab/blob/main/DATASETS.md
I downloaded it from this link.

Ran12256, 2025-03-26:
> I have not been able to get the code to work correctly yet, so I don't know if it is correct, https://github.com/facebookresearch/habitat-lab/blob/main/DATASETS.md I downloaded it from this link.

Have you solved the problem? I met the same problem.